espruino/Espruino #1988: Variable value changed automatically

vidalouiswang, 2021-02-23:
Here is the code:
`rtnValue.reply = function(data) {
            print("data2:" + data);
            print("into the reply");
            this.t2 = to36(this.t2);
            print("t2:" + this.t2);
            var hash = sha1(this.user.password + this.t2);
            hash = to36(hash, 8);
            print("hash:" + hash);
            var p = 0;
            var encrypted = "";
            var hash2 = hash;

            {
                let confusion = sha1(this.t2 + this.user.password);
                for (let i of confusion) { p = p ^ i.charCodeAt(0); }
                print("p:" + p);
                for (let i of data) { encrypted += String.fromCharCode(p ^ i.charCodeAt(0)); }
                print("data:" + data);
                print("res:" + encrypted);
                encrypted = btoa(encrypted);
                print("res2:" + encrypted);
            }
            data = null;
            var content =
                this.from +
                id.short +
                String.fromCharCode(this.user.username.length) +
                String.fromCharCode(this.t2.length) +
                this.user.username +
                this.t2 +
                hash2 +
                encrypted;

            socket.send(content);
        };
`

prints:
[Jtarget:HHHHHHHH
from:2c1pd6of
userLen:8
tLen:8
user:username
t:klia831w,1614101275508
hash:2gjoa2zx
user exist
localhash:2gjoa2zx
authorized
data2:online
into the reply
t2:181lg
hash:ai9qwh4w
p:13
data:9
res:4
res2:NA==

The argument of this function "data" value changed automatically, but there is not clause can do that, why?
And also you can see the "hash" and "hash2", for same reason I made a copy named "hash2" from hash.

gfwilliams, 2021-02-24:
* What board are you running on?
* What arguments to do call `reply` with?
* What is `to36` and `sha1` and `this.user` ?

If we're going to be able to help you at all, we need you to supply code that actually runs. 

I just tried the following in the emulator and it works fine for me:

```
var rtnValue = {user : {password:"xyz",username:"x"}};
var id = {short:1};
rtnValue.reply=function(data) {
            print("data2:" + data);
            print("into the reply");
            this.t2 = to36(this.t2);
            print("t2:" + this.t2);
            var hash = sha1(this.user.password + this.t2);
            hash = to36(hash, 8);
            print("hash:" + hash);
            var p = 0;
            var encrypted = "";
            var hash2 = hash;

            {
                let confusion = sha1(this.t2 + this.user.password);
                for (let i of confusion) { p = p ^ i.charCodeAt(0); }
                print("p:" + p);
                for (let i of data) { encrypted += String.fromCharCode(p ^ i.charCodeAt(0)); }
                print("data:" + data);
                print("res:" + encrypted);
                encrypted = btoa(encrypted);
                print("res2:" + encrypted);
            }
            data = null;
            var content =
                this.from +
                id.short +
                String.fromCharCode(this.user.username.length) +
                String.fromCharCode(this.t2.length) +
                this.user.username +
                this.t2 +
                hash2 +
                encrypted;

            socket.send(content);
        };
function to36(x) { return x; }
function sha1(x) { return x; }
rtnValue.reply("online")
```

vidalouiswang, 2021-02-24:
> * What board are you running on?
> * What arguments to do call `reply` with?
> * What is `to36` and `sha1` and `this.user` ?
>
> If we're going to be able to help you at all, we need you to supply code [...]

Espruino running on the NodeMCU 8266 4MB, the argument 'data' is a string,
data = String.fromCharCode(0) + "online";
this.user = { username: "testuser", password: "27d5482eebd075de44389774fce28c69f45c8a75" }；
Free memory more than 500 blocks when running this function.
And here is the code:
`
sha1 = origin => {
            let src = require("crypto").SHA1(origin);
            let res = "";
            for (let i of src) res += i < 16 ? ("0" + i.toString(16)) : i.toString(16);
            return res;
        };

to36 = (data, len) => {
            if (typeof data == "number") {
                data = data.toString(36);
            } else {
                //only for hex
                if (/[^0-9a-fA-F]/.test(data)) { return; }
                //try {
                //对齐16进制
                if (data.length % 2) { data = "0" + data; }
                //16进制位数较多时，在不同的平台可能会结果不一样
                //所以先进行转换，此场景默认16进制为hash
                if (data.length > 16) {
                    data = data.substr(0, 6) + data.substr(data.length - 6, 6);
                }
                data = parseInt(data, 16);

                data = data.toString(36);
                //} catch (e) { print(e); return; }
            }

            if (len) {
                if (data.length > len) {
                    //限制输出长度
                    return data.substr(0, len);
                } else if (data.length < len) {
                    //长度不够，需要补齐
                    let bound = len - data.length;
                    while (bound--) {
                        data = sf(0) + data;
                    }
                    return data;
                }
            }
            return data;
        }
`

vidalouiswang, 2021-02-24:
> * What board are you running on?
> * What arguments to do call `reply` with?
> * What is `to36` and `sha1` and `this.user` ?
>
> If we're going to be able to help you at all, we need you to supply code [...]

In the front of this function, there no any clause changed the value of data, but it did changed.
I modified the module websocket, when recieved data from server and after unmasked,
the original clause is `this.emit("message",g);`, I changed it to `this.decode(g);`.
I added a function to the builtin module:
`
g.prototype.decode  = function(msg){
         //返回值
    var rtnValue = { err: null, data: "" };

    //检查消息体
    if (!msg) { rtnValue.err = "empty"; }
    if (msg.length < 8) { rtnValue.err = "short length"; }

    var detach = len => {
        len = len || 8;
        var a = msg.substr(0, len);
        msg = msg.substr(len);
        return a;
    };

    rtnValue.target = detach();
    rtnValue.from = detach();
    rtnValue.userLen = detach(1);
    rtnValue.tLen = detach(1);
    rtnValue.user = detach(rtnValue.userLen.charCodeAt(0));
    rtnValue.time = detach(rtnValue.tLen.charCodeAt(0));
    rtnValue.hash = detach();
    var src = detach(256);
    print("target:" + rtnValue.target);
    print("from:" + rtnValue.from);
    print("userLen:" + rtnValue.userLen.charCodeAt(0));
    print("tLen:" + rtnValue.tLen.charCodeAt(0));
    print("user:" + rtnValue.user);
    print("t:" + rtnValue.time + "," + from36(rtnValue.time));
    print("hash:" + rtnValue.hash);

    rtnValue.broadcast = rtnValue.target == "HHHHHHHH";

    //查询用户
    rtnValue.user = users(rtnValue.user || " ");

    //认证
    rtnValue.authorized = false;
    if (rtnValue.user) {
        print("user exist");
        let localHash = sha1(rtnValue.user.password + (rtnValue.time || 1));
        localHash = to36(localHash, 8);
        print("localhash:" + localHash);

        rtnValue.authorized = localHash == (rtnValue.hash || " ");
    } else {
        rtnValue.err = "user not exist";
    }

    if (rtnValue.authorized) {
        print("authorized");
        rtnValue.t2 = time;
        var socket = this;

        //回复
        rtnValue.reply = function(data) {
            print("data2:" + data);
            print("into the reply");
            this.t2 = to36(this.t2);
            print("t2:" + this.t2);
            var hash = sha1(this.user.password + this.t2);
            hash = to36(hash, 8);
            print("hash:" + hash);
            var p = 0;
            var encrypted = "";
            var hash2 = hash;

            {
                let confusion = sha1(this.t2 + this.user.password);
                print("data6:" + data);
                print("mem:" + mem);
                for (let i of confusion) { p = p ^ i.charCodeAt(0); }
                print("data8:" + data);
                print("p:" + p);
                for (let i of data) { encrypted += String.fromCharCode(p ^ i.charCodeAt(0)); }
                print("res:" + encrypted);
                encrypted = btoa(encrypted);
                print("res2:" + encrypted);
            }
            data = null;
            var content =
                this.from +
                id.short +
                String.fromCharCode(this.user.username.length) +
                String.fromCharCode(this.t2.length) +
                this.user.username +
                this.t2 +
                hash2 +
                encrypted;

            socket.send(content);
        };

        if (rtnValue.broadcast) {
            if (rtnValue.t2 - (val("broadcast") || 1) > 3000) {
                val("broadcast", rtnValue.t2);
                rtnValue.reply(String.fromCharCode(0) + "online");
            }
        } else {
            print(rtnValue.target);
            if (rtnValue.target == id.short) {
                let confusion = sha1(rtnValue.time + rtnValue.user.password);
                let p = 0;
                for (let i of confusion) { p = p ^ i.charCodeAt(0); }

                rtnValue.data = "";
                for (let i of src) {
                    rtnValue.data += String.fromCharCode(i.charCodeAt(0) ^ p);
                }

                src = null;
                rtnValue.contentType = rtnValue.data.charCodeAt(0);
                rtnValue.data = rtnValue.data.substr(1);

            } else {
                rtnValue.err = "not me";
            }
        }
    } else {
        rtnValue.err = "unauthorized";
    }

    this.emit("res", rtnValue);
}
`

vidalouiswang, 2021-02-24:
> * What board are you running on?
> * What arguments to do call `reply` with?
> * What is `to36` and `sha1` and `this.user` ?
>
> If we're going to be able to help you at all, we need you to supply code [...]

id is an json object of global,
id = {hex: sha1(getSerial())};
id.short  = to36(id.he , 8);

gfwilliams, 2021-02-24:
Can you post up **A SINGLE SECTION OF CODE** that I can just run in the IDE and reproduce the problem?

I don't believe any of the code you pasted up will 'just work'.

I'm trying to help, but this is **free software**. I'm trying to help you **for free**.

I don't have time to fix the formatting and cut and paste all the different sections of code you've just supplied

vidalouiswang, 2021-02-25:
> Can you post up **A SINGLE SECTION OF CODE** that I can just run in the IDE and reproduce the problem?
>
> I don't believe any of the code you pasted up will 'just work'.
>
> I'm trying to help, but this [...]

The code:

fs = require("Storage");
fs.eraseAll();
fs.compact();
E.reboot();
fs = require('Storage');
fs.write(`.boot0`, `g=global;g.emit=((e,t)=>{if(g["#"+e])for(let r of g["#"+e])r.apply(g,[t])});g.on=((e,t)=>{g["#"+e]=g["#"+e]||[];g["#"+e].push(t)});{let e=Object.defineProperty;e(g,"time",{get:e=>Math.round((new Date).getTime())});e(g,"mem",{get:e=>process.memory().free})}fs=require("Storage");g.sf=String.f=String.fromCharCode;loadModule=((e,t,r,i,n)=>{i=i||global;r=r||[];let a;m=require(e);t?m[t]&&(a=m[t].apply(i,r)):m.apply&&(a=m.apply(i,r));n||Modules.removeCached(e);return a});writeFile=fs.write;readFile=fs.read;writeJS`, 0, 2293);
fs.write(`.boot0`, `ON=fs.writeJSON;readJSON=(e=>{try{return fs.readJSON(e)}catch(e){}});{function DB(){this.name="db";readJSON(this.name)||writeJSON(this.name,{});this.queue=[];this.cache=[];this.running=!1}DB.prototype.findInArray=function(e,t){return t?this.cache.find(t=>t.key==e):this.queue.find(t=>t.key==e)};DB.prototype.innerUpdate=function(e,t,r,i){var n=this.findInArray(e,i);if(n){n.value=t;i||(n.t=time)}else{n={key:e,value:t,t:r?0:time};this.queue.push(n)}this.start()};DB.prototype.update=function(e,t,r){this.innerUpd`, 512);
fs.write(`.boot0`, `ate(e,t,r);this.innerUpdate(e,t,r,!0)};DB.prototype.start=function(){if(!this.running){this.timer&&clearInterval(this.timer);this.running=!0;this.timer=setInterval(()=>{var e=time,t=this.queue.filter(t=>e-t.t>=6e3);if(t.length){let r=readJSON("db")||{};for(let e of t)r[e.key]=e.value;writeJSON("db",r);this.queue=this.queue.filter(t=>e-t.t<6e3);if(!this.queue.length){clearInterval(this.timer);this.running=!1}}},1e3)}};DB.prototype.innerBuildCache=function(e,t){var r=this.findInArray(e,!0);r?r.times++:(t||"st`, 1024);
fs.write(`.boot0`, `ring"==typeof t)&&this.cache.push({key:e,value:t,times:1});return t};DB.prototype.clearCache=function(e){e=e||10;if(this.cache.length>e){var t={times:8e8};for(let e of this.cache)e.times<t.times&&(t=e);if(t.hasOwnProperty(key)){this.cache=this.cache.filter(e=>e.key!=t.key);return 1}}};DB.prototype.read=function(e){if(e){var t=this.findInArray(e,!0)||this.findInArray(e);if(t){this.innerBuildCache(e);return t.value}{let t=readJSON("db");if(t.hasOwnProperty(e))return this.innerBuildCache(e,t[e])}}};g.db=new DB`, 1536);
fs.write(`.boot0`, `}g.valuer=loadModule("valuer0");val=((e,t,r)=>{if(!t&&"string"!=typeof t)return db.read(e);db.update(e,t,r)});val("entry",val("entry")||"0");E.reboot2=(e=>{g.emit("reboot");setTimeout(()=>{E.reboot()},3e3)});eval(readFile("entry"+val("entry")));`, 2048);
E.reboot();
val('wifissid','Vida',true)
val('wifipass','3W9%5x.V2_4d,X5$',true)
fs.write(`io0`, `function io(t,e){let o=this;o.socket=null;e=e||{};o.host=t;o.port=e.port||80;o.protocolVersion=e.protocolVersion||13;o.origin=e.origin||"Espruino";o.keepAlive=1e3*e.keepAlive||6e4;o.masking=void 0===e.masking||e.masking;o.path=e.path||"/";o.protocol=e.protocol;o.lastData="";let i=t=>Math.random().toString(36).substr(2,8),r=btoa(i()+i()),n=r+"258EAFA5-E914-47DA-95CA-C5AB0DC85B11";o.key={source:r,hashed:btoa(loadModule("crypto","SHA1",[n]))};o.dbl=sf(13)+sf(10);o.headers=e.headers||{}}io.prototype.initialize=`, 0, 2435);
fs.write(`io0`, `function(){let t=this;require("net").connect({host:t.host,port:t.port},t.onConnect.bind(t))};io.prototype.onConnect=function(t){let e=this;e.socket=t;t.on("data",e.parseData.bind(e));t.on("close",t=>{e.emit("close")});e.handshake(e)};io.prototype.parseData=function(t){var e=this;this.emit("rawData",t);this.lastData.length&&(t=this.lastData+t,this.lastData="");t.length>1024&&(t="",this.emit("error","long content"));if(this.connected)for(;t.length;){var o=2,i=15&t.charCodeAt(0),r=127&t.charCodeAt(1);if(126==r`, 512);
fs.write(`io0`, `)r=t.charCodeAt(3)|t.charCodeAt(2)<<8,o+=2;else if(127==r){t="";return}var n=r+o+(128&t.charCodeAt(1)?4:0);if(n>t.length){this.lastData=t;break}switch(i){case 1:i=[0,0,0,0];128&t.charCodeAt(1)&&(i=[t.charCodeAt(o++),t.charCodeAt(o++),t.charCodeAt(o++),t.charCodeAt(o++)]);for(var a="",s=0;s<r;s++)a+=String.fromCharCode(t.charCodeAt(o++)^i[3&s]);this.decode.call(e,a)}t=t.substr(n)}else if(t.includes(e.key.hashed)&&t.includes(e.dbl+e.dbl)){t=t.substr(t.indexOf(e.dbl+e.dbl)+4);e.connected=!0;e.emit("open");glob`, 1024);
fs.write(`io0`, `al.emit("online");e.lastData=t}};io.prototype.handshake=(t=>{let e=t.dbl,o="GET "+t.path+" HTTP/1.1"+e+"Host: "+t.host+e+"Upgrade: websocket"+e+"Connection: Upgrade"+e+"Sec-WebSocket-Key: "+t.key.source+e+"Sec-WebSocket-Version: "+t.protocolVersion+e+"Origin: "+t.origin;t.socket.write(o+e+e)});io.prototype.decode=function(t){require("decode0").decode.call(this,t)};io.prototype.send=function(t,e,o){let i=this.socket;if(i){i.write(sf(void 0===e?129:e,(t.length>125?126:t.length)+128));t.length>125&&(i.write(sf`, 1536);
fs.write(`io0`, `(t.length>>8)),i.write(sf(t.length)));if(o){let e="";{let o=[];for(let t=0;t<4;t++){o[t]=Math.floor(255*Math.random());e+=sf(o[t])}for(let i=0;i<t.l;i++)e+=sf(t.charCodeAt(i)^o[3&i]);t=null}i.write(e)}else i.write(sf(0,0,0,0)+t)}else this.emit("error",108)};io.prototype.close=function(){try{this.socket.end()}catch(t){}};exports=function(t,e){var o=new io(t,e);o.initialize();return o};`, 2048);
[... cut ...]
fs.write(`entry0`, `);net.on("res",e=>{if(e.err)print("err:"+e.err);else{log("data:");print(e.data)}})}})}else log("no available wifi info");fs.write("*entry","1");val("avaEntry",val("entry"));log("entry initialized")})();`, 2048);
E.reboot();

Look, I don't expect espruino can make any product for sell now, just like js and would like to see this project become popular.
So I built some simple code for test, try to help espruino become robust, my work is also for free, it's only for this project.

closed 2021-02-25 by vidalouiswang

vidalouiswang, 2021-06-13:
> Can you post up **A SINGLE SECTION OF CODE** that I can just run in the IDE and reproduce the problem?
>
> I don't believe any of the code you pasted up will 'just work'.
>
> I'm trying to help, but this [...]

I apologize to Gordon Williams. I am sorry for those words. Indeed, I should provide minimal code snippet for project issue. Hope you won't mind.

gfwilliams, 2021-06-14:
Thanks for the reply! No problem at all :)

vidalouiswang, 2021-06-14:
> Thanks for the reply! No problem at all :)

You are such a nice man and you are really a gentleman worthy of being respected. 👍