espruino/Espruino #1990: puckjs device connection limit

bettse, 2021-03-06:
Is the puckjs limited to connecting to one peripheral at a time?  Any difference between the v1 and v2?  How difficult would this be to increase (if I were motivated to write some C)?  

(I also had sent you this question over pateon, but realized that might not be something you check)

gfwilliams, 2021-03-08:
Hi! Just replied on Patreon.

There's no difference in the microcontroller capabilities between v1 and v2, and this is definitely possible. There's actually in issue filed for it on https://github.com/espruino/Espruino/issues/1360 and it's something I've been meaning to take a look at.

Basically, it shouldn't be *too* hard. The gotcha is that right now all the functions expect just one connection so they just look in a single variable for the connection handle - they need to be modified to check inside the JS object for a 'connection handle' so they can do the right thing on the right connection.

Finally you just need to tell the softdevice you want 2(or more) active connections, and change Espruino's RAM usage accordingly. One reason I haven't been prioritising this is that to enable this it uses up RAM constantly, so regardless of whether you need 2 connections or not, you end up with less memory available all the time.

Still, I think it could make sense as a separate firmware image...

gfwilliams, 2021-03-08:
Closing this though, as it's basically a duplicate of #1360

closed 2021-03-08 by gfwilliams