espruino/Espruino #1992: Parser incorrectly returning syntax error

GaryOtt, 2021-03-17:
There is a bug that prevents the below ugly syntax from parsing correctly. I have encountered this problem quite a few times over the last couple of years but it has been very difficult to identify the problem when digging through minified code. Today I was fortunate to encounter the problem in a script that minified to only 519 bytes. It is something of a relief to finally be able to share a minimalist example of the issue. 

```
 ____                 _ 
|  __|___ ___ ___ _ _|_|___ ___ 
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v08.191 (c) 2019 G.Williams

Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate

>var e = { encode: ()=> {}};
={ 
  encode: function (undefined) {}
 }
>r=>e.encode(
:(e=>{})(r)
:);
Uncaught SyntaxError: Got '}' expected EOF
 at line 2 col 3
(e=>{})(r)
  ^
 at line 2 col 6
(e=>{})(r)
     ^
>
```

The expected output would be more like this output from NodeJS:

```
Garys-MacBook-Pro-2:~ gary$ node
Welcome to Node.js v12.18.1.
Type ".help" for more information.
> var e = { encode: ()=> {}};
undefined
> 
> r=>e.encode(
... 	(e=>{})(r)
... );
[Function]
> 

```

gfwilliams, 2021-03-17:
Thanks for narrowing this down!

I don't have time to look at it and fix it right this second, but it seems it can be narrowed down even more:

```
// fails
r=>print( (e=>{})(r) )
// works
r=>print( (()=>{})(r) )
```

gfwilliams, 2021-04-16:
Ok, looks like:

```
if(0)e=>{}
```

will do it - the issue is actually parsing the arrow function when Espruino isn't executing.

In `jspeExpressionOrArrowFunction`, `jspeAssignmentExpression` returns 0 when `JSP_SHOULD_EXECUTE` is false, which causes it to fail.

It's actually `jspeFactor` that's at fault. Pushing a fix now.

commit 8c51e539 (2021-04-16): Fix issue parsing `r=>print((e=>{})(r))` (fix #1992)
closed 2021-04-16 by gfwilliams