espruino/Espruino #1994: Add more Cipher Suites to Espruino

andreykotomin, 2021-03-30:
Hello!
Can additional cipher suites be added to Espruino? I did a little research and figured out how to do it:
[http://forum.espruino.com/conversations/361833/](url)
It'd be useful if Espruino could access modern https web-sites and APIs.

gfwilliams, 2021-04-06:
Thanks for this - yes, this looks like a good thing to add. The only gotcha is ensuring it's not going to cause compilation problems for other targets before enabling as it could use more flash - but when I get a moment I'll check on that.

It seems you are significantly better versed in this than me: So are there any cipher suites that Espruino supports that are currently deprecated/unused and could be *removed* from the build?

andreykotomin, 2021-04-06:
Unfortunately I'm not an expert in cyber security, however I've found this article with some statistics about most used cipher suites:
https://scotthelme.co.uk/top-1-million-analysis-march-2020. It seems that the TLS_RSA suites are hardly used anymore.

gfwilliams, 2021-04-19:
Just a note on this:

> MBEDTLS_KEY_EXCHANGE_DHE_RSA_ENABLED, MBEDTLS_DHM_C, MBEDTLS_GCM_C, and also added two lines:
> libs/crypto/mbedtls/library/dhm.c \
> libs/crypto/mbedtls/library/gcm.c

I've checked here and for Espruino WiFi:

* before (just RSA) 341616
* as you suggested 347120
* as you suggested with RSA removed 346208

So it'd save about 1k taking the old one out. I'm just checking whether everything still builds ok, but if it does I think it makes sense to leave both in for now to avoid breaking stuff that might have worked previously.

commit 45f9f26c (2021-04-19): Add DHE RSA key exchange for TLS/HTTPS (fix #1994)
closed 2021-04-19 by gfwilliams