espruino/Espruino #1996: Latest GCC gives `section .noinit will not fit in region NOINIT`

masoncj, 2021-04-10:
Hello! When compiling latest Espruino (1da552100c41936f0cf7be7d4a8305a88153a9f3) with `gcc-arm-none-eabi-10-2020-q4-major` on MacOS X 11.2.3, I get:

```
make clean && DFU_UPDATE_BUILD=1 BOARD=PUCKJS RELEASE=1 make
...
/Applications/ARM/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: espruino_2v08.208_puckjs.elf section `.noinit' will not fit in region `NOINIT'
/Applications/ARM/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: region `NOINIT' overflowed by 4 bytes
```

Any suggestions? Which version of the toolchain are you guys using?

masoncj, 2021-04-12:
(Part of my motivation for using this version of GCC is that it seems to be the first major release that ships as a signed and notarized binary for MacOS.  I haven't yet figured out how to convince Big Sur to run unsigned versions of GCC, despite trying several work arounds described in various places.  I am unable to convince it to allow `as` to run.)

masoncj, 2021-04-12:
If I edit `targetlibs/nrf5x_12/nrf5x_linkers/linker_nrf52_ble_espruino_bootloader.ld` to have:

```
NOINIT (rwx) :  ORIGIN = 0x2000fff0, LENGTH = 20
```

It seems to at least successfully link.  However when I DFU it to the Puck.js, the Web IDE is no longer able to connect.

I'm getting the sense that to do what I want to do (add low-level nRF52 functions to Espruino) I'm going to need to figure out a way to attach SWD to the Puck or use a different board for development that has a DAP interface for OpenOCD or similar.  So I can at least tell what's going on.

gfwilliams, 2021-04-12:
Ok, thanks - it'd be interesting if you could `make lst`, look at the list file, and see what's taking up all the space in `NOINIT`? There should only be a few things, but maybe the default alignment has changed.

You could always just spin up a Linux VM and do compilation in that - it'd be much easier to keep the same compiler as we use - then you're not worrying about whether things aren't working because of your changes, or the different compiler :)

In terms of development, you can do uploads wirelessly but it's slow and painful. I'd definitely recommend adding an SWD interface. You can just get a NRF52DK and either [we have a jig](https://shop.espruino.com/puckjs-jig) or you can get away with just 3 wires soldered to the test points.

masoncj, 2021-04-13:
Here are the lines from the list file that match `noinit`:

```
  4 .noinit       00000014  2000fff0  2000fff0  0006fff0  2**3
2000fff0 l    d  .noinit	00000000 .noinit
20010000 l     O .noinit	00000004 lastSystemTimeInv
2000fff0 g     O .noinit	00000004 .hidden lastSystemTime
2000fff8 g     O .noinit	00000008 .hidden baseSystemTime
2000fff0 g       .noinit	00000000 __noinit_start__
20010004 g       .noinit	00000000 __noinit_end__
```

This only seems to add up to 16, so I'm a bit confused.

I've ordered a Nordic 52832 DK and one of the [Makerdiary ones](https://wiki.makerdiary.com/nrf52832-mdk/).  I also have a Jlink around somewhere that I can probably wire up.

gfwilliams, 2021-04-13:
Thanks! it's actually the first column (the address) that shows the issue - it seems that version of GCC is using 8 byte alignment (not 4).

While we could tag lastSystemTime/etc with an attribute asking for 4 byte alignment, I think is GCC is applying a blanket 8 byte alignment policy it's going to cause problems elsewhere as well (probably why your build didn't work). Maybe you could have a quick google and see if there's a way to globally force alignment for 4 bytes on GCC10?

fanoush, 2021-04-15:
Saw this too on linux gcc. BTW why this noinit is not simply put put after .bss section? it is used only for brief moment when rebooting from same firmware to same firmware so the location does not need to be hardcoded. Also current way needs  to rebuild all bootloaders for all boards to use this feature (and wastes 16 bytes forever). Also there could be some simple checksum to verifiy the data in whole structure/setion correct, seen date restored to some nonsense. With checksum and movement after .bss we could put more stuff there that would be kept over reboot before JS variables are available (I'm still thinking about e.g. keeping MTU there and sizing JS variables at boot time according to that - they could be put right from the beginning of data segment  to 0x2000000 and then at boot time it could skip stuff needed to softdevice and start right after it)

fanoush, 2021-04-15:
if there is no strong reason to keep it above stack I can try it, I already tried to put jsvars at the beginning like that
```
diff --git a/targetlibs/nrf5x_12/nrf5x_linkers/gcc_nrf5x_espruino_common.ld b/targetlibs/nrf5x_12/nrf5x_linkers/gcc_nrf5x_espruino_common.ld
index cb78aede..f3a46d7b 100644
--- a/targetlibs/nrf5x_12/nrf5x_linkers/gcc_nrf5x_espruino_common.ld
+++ b/targetlibs/nrf5x_12/nrf5x_linkers/gcc_nrf5x_espruino_common.ld
@@ -91,6 +91,17 @@ SECTIONS
   _etext = .;
   _sidata = .;
 
+  .jsvars (NOLOAD):
+  {
+    . = ALIGN(4);
+    __jsvars_start__ = .;
+    _sjsvars = .;
+    *(.jsvars*)
+    . = ALIGN(4);
+    __jsvars_end__ = .;
+    _ejsvars = .;
+  } > RAM
+
   .data : AT (__etext)
   {
     __data_start__ = .;
```

so same thing can be done with `.noinit` and can be put right after .bss in that same file 
```
  .bss :               
  {                    
    . = ALIGN(4);      
    __bss_start__ = .; 
    _sbss = .;         
    *(.bss*)           
    *(COMMON)          
    . = ALIGN(4);
    __bss_end__ = .;
    _ebss = .;
  } > RAM

```

I guess not restoring that data when upgrading to new FW is not an issue.

gfwilliams, 2021-04-15:
Actually sticking at the end of stack sounds good - I just want somewhere that doesn't get cleared either at reboot or by the bootloader (and I believe on Bangle.js the bootloader leaves the RAM alone there anyway?).

I was under the impression that the stack got cleared to 0 at boot though?

> Also there could be some simple checksum to verifiy the data in whole structure/section is correct, seen date restored to some nonsense. 

That already exists if if you look (the value is stored normally and then inverted) - I think you should only have seen the dodgy date issue for a few days before I fixed it.

fanoush, 2021-04-15:
> just want somewhere that doesn't get cleared either at reboot or by the bootloader

oh, right, the bootloader, even if it does not enter DFU it might clear it if it's .data+.bss section is larger than espruino's, will test. But true that it looks a bit fragile. Just  don't like wasting that space forever when ram is so tight and it is not needed during runtime.

I don't think stack is cleared, only .bss section is https://github.com/espruino/Espruino/blob/master/targetlibs/nrf5x_12/components/toolchain/gcc/gcc_startup_nrf52.S#L372 , will test. I think I previously needed to paint stack with some pattern myself to know how low it usually goes since it was not cleared.

gfwilliams, 2021-04-15:
> and it is not needed during runtime.

That's not entirely true actually - `lastSystemTime/baseSystemTime` had to exist already, so the only extra is 4 bytes for `lastSystemTimeInv`.

But if there's a way to stick them at the top of the stack and they won't get cleared then that seems preferable if only because it means we don't need any linker hacks.

fanoush, 2021-04-15:
Oh got it. Thanks for explanation. So it is same variable as before, just moved there. OK, makes sense now. So then I was after something a bit different - temporary storage over reboot. And yes bottom of stack could be used for that.

I think stack size is not formally set anywhere in espruino(?) so there is no symbol for that but it grows down till it hits end of static variables  - LINKER_END_VAR = stackEndAddress
https://github.com/espruino/Espruino/blob/master/src/jswrap_process.c#L189 So putting some `struct bootVars *vars = &LINKER_END_VAR` there with crc16, lastSystemTime and possibly other stuff should be possible. This could be then passed to (or used in) various js[xx]Init methods to override some low level defaults (when jsvars or flash storage is not practical or available yet). However now I indeed see that it is something quite different than you did with keeping system time.

 So is getting rid of this linker hack with .noinit section worth creating something like that and keep copy of system time there?

fanoush, 2021-04-16:
I tried to add structure with checksum that is preserved across reboots
https://gist.github.com/fanoush/b4897b821b4e69f18f18e1cf8ef69cf3
I also used it to preserve app_ram_base which can be used to resize variables at boot time if softdevice needs more memory - you you may ignore that part. The interesting bits start here https://gist.github.com/fanoush/b4897b821b4e69f18f18e1cf8ef69cf3#file-bootparams-diff-L150
it works but I am not sure about it. It shows disadvantage or this approach -  one must copy the data to/from the structure at some point in time which is extra code. Your approach of having those variables stored directly in some .noinit section skips this step so is more straightforward - the only issue is the checksum. However with some clever linker script marking start/end of the data the checksum could be done over whole .noinit section. Also if checksum would not match the section could be automatically zeroed like bss.

So now I think better idea is to move .noinit section to end of static data/bottom of stack but otherwise keep this section and some linker script magic to checksum and clear it. Advantage is that sizing and location would be automatic and it would not need extra absolute memory area defined.

gfwilliams, 2021-04-16:
Thanks! Can you not just write directly to `bootParams->baseSystemTime`?

The issue with writing to a separate var and copying is that in the case of something like a watchdog reset (long press BTN1+2 on Bangle.js), hardfault, or anything other than a user-requested reboot the time would get lost.

> So now I think better idea is to move .noinit section to end of static data/bottom of stack but otherwise keep this section and some linker script magic to checksum and clear it.

That sounds like a good plan to me... And no real big changes... So where does NOINIT sit at the moment?

I figured the bootloader itself didn't need any changes for this work at the moment - is that not the case?

masoncj, 2021-04-16:
Are we still concerned there might be more global alignment issues with GCC10? I did spend some time poking around the GCC release notes to see if they made an alignment change — and for a global option to reverse it — and didn’t find much.  I can keep hunting though.

fanoush, 2021-04-16:
> Thanks! Can you not just write directly to `bootParams->baseSystemTime`?

use it everywhere as primary storage? then it is like the .noinit section but with ugly syntax. Marking variable with .noinit attribute and then using it  by simple name looks nicer and is basically same then.

>  And no real big changes...

yes, and the watchdog issue is also true so keeping a copy is not good indeed

>  So where does NOINIT sit at the moment?

Not sure what you mean exactly but In my patch the structure sits above static variables/bottom of stack - the `&LINKER_END_VAR` which you call stackEndAddress here https://github.com/espruino/Espruino/blob/master/src/jswrap_process.c#L189

the .noinit section could be there too so the layout would be .data (=initialized data) then .bss (=zeroed data) then .noinit (=not zeroed not initialized)

> I figured the bootloader itself didn't need any changes for this work at the moment - is that not the case?

Yes, you're right, it is true for SDk12 espruino bootloader, it uses just half of the ram so the issue is not there. There are other bootloaders in other devices that may have issue with it but not the espruino one. Also later Nordic sdks has it enlarged to full 64K e.g. 
`nRF5_SDK_14.2.0_17b948a/examples/dfu/bootloader_secure_ble/pca10040/armgcc/secure_dfu_ble_gcc_nrf52.ld` has `RAM (rwx) :  ORIGIN = 0x20002830, LENGTH = 0xd7d0` or sdk15 too `nRF5_SDK_15.0.0_a53641a/nRF5_SDK_15.0.0_a53641a/example/dfu/secure_bootloader/pca10040_ble/armgcc/ecure_bootloader_gcc_nrf52.ld` - `RAM (rwx) :  ORIGIN = 0x200057b8, LENGTH = 0xa848`.

In espruino tree there is nrf52840 one here https://github.com/espruino/Espruino/blob/master/targetlibs/nrf5x_15/nrf5x_linkers/secure_bootloader_gcc_nrf52.ld#L9 but it is probably not used for anything now.

fanoush, 2021-04-16:
> Are we still concerned there might be more global alignment issues with GCC10? 

I saw no other issue, when enlarged from 16 to 24, it linked and then worked just fine for me. Didn't see any crashes or hangs. Oh, when seeing your `NOINIT (rwx) :  ORIGIN = 0x2000fff0, LENGTH = 20` now - it is obviously wrong as there is no RAM in last 4 bytes, you would need to move it down to add more - like `ORIGIN = 0x2000fff0 - 4, LENGTH = 16 + 4`

gfwilliams, 2021-04-16:
Ahh - that might have caused the crash I guess.

But the alignment issue is strange - especially as it seems the alignment was actually set already:

https://github.com/espruino/Espruino/blob/master/targets/nrf5x/jshardware.c#L99

Perhaps GCC 10 doesn't like the way the attribute was specified. Any thought @fanoush ? Or maybe the linker file could specify an alignment of 4?

fanoush, 2021-04-16:
> Perhaps GCC 10 doesn't like the way the attribute was specified. Any thought @fanoush ? Or maybe the linker file could specify an alignment of 4?

Linker file seems to already specify it, I tried various additional alignment syntax on the section in linker file like
```
  .noinit (NOLOAD) : ALIGN(4) SUBALIGN(4)
  {
    . = ALIGN(4);
```
also raised it to 8 and 16 since at  0x2000fff0 it already is aligned to 16 bytes and nothing helped. I guess the issue is with the alignment of that variable itself. And for some unknown reason compiler does not place it to be the first variable in the section(?) 
Also looks like it is not possible to change default alignment of some type  easily since it could break stuff https://gcc.gnu.org/legacy-ml/gcc-help/2009-01/msg00091.html. That JsSysTime type is int64_t which seems to be aligned to 8 bytes on other platforms too  so maybe they fixed it for gcc 10. In fact I noticed they also fixed some linking bugs with weak symbols I had earlier with nordic examples - I needed to reorder sources in makefiles to link interrupt handlers properly. Those with weak symbols needed to go first like [here](https://gist.github.com/fanoush/3420b9a3bccaaff3432ecf97f707958d#file-sdk14-2-dfutarg-lf_rc-diff-L322). This is no longer needed in gcc 10.

BTW your last commit fixes this for me, works in gcc10 now.

masoncj, 2021-04-16:
Nice!  

While the standard Puck.js code doesn't seem to fit, 

```
Testing espruino_2v08.220_puckjs.elf for PUCKJS
STORAGE: 442368 -> 483328
FS DATA: 442428 -> 442444 (16 bytes)
CODE: 126976 -> 442444 (314980 bytes)
==========================
 CODE AND STORAGE OVERLAP
   by 76 bytes
==========================
```

The minimal version compiles and works! 

Thanks so much for the fix, and for the explanation about how to read and think about the linker output. I'm still learning.

fanoush, 2021-04-18:
> While the standard Puck.js code doesn't seem to fit,

maybe you're not building with RELEASE=1 ?

gfwilliams, 2021-06-18:
I believe this is fixed now

closed 2021-06-18 by gfwilliams