espruino/Espruino #2000: advertising broken with S140 >= 6.1.0

fanoush, 2021-04-21:
with 6.0.0 it works but with 6.1.0 and up no device is visible. Found the issue and solution described here

https://devzone.nordicsemi.com/f/nordic-q-a/37473/advertising-does-not-start-with-s140-6-1-0-same-hex-run-without-issue-with-s140-6-0-0/145157#145157

without understanding details of espruino advertising data lifecycle, quick fix like this seems to work (= device is visible so I can at least connect to console over bluetooth repeatedly)
```
diff --git a/targets/nrf5x/bluetooth.c b/targets/nrf5x/bluetooth.c
index b667c0ff..1d9815da 100644
--- a/targets/nrf5x/bluetooth.c
+++ b/targets/nrf5x/bluetooth.c
@@ -2513,7 +2513,7 @@ uint32_t jsble_advertising_start() {
   adv_params.interval = bleAdvertisingInterval;

   uint32_t err_code = 0;
-  uint8_t m_enc_scan_response_data[31]; // BLE_GAP_ADV_SET_DATA_SIZE_MAX
+static  uint8_t m_enc_scan_response_data[31]; // BLE_GAP_ADV_SET_DATA_SIZE_MAX
   uint16_t m_enc_scan_response_data_len = sizeof(m_enc_scan_response_data);
 #if NRF_SD_BLE_API_VERSION<5
   err_code = adv_data_encode(&scanrsp, m_enc_scan_response_data, &m_enc_scan_response_data_len);
@@ -2528,8 +2528,10 @@ uint32_t jsble_advertising_start() {

   //jsiConsolePrintf("adv_data_set %d %d\n", advPtr, advLen);
 #if NRF_SD_BLE_API_VERSION>5
-  ble_gap_adv_data_t d;
-  d.adv_data.p_data = (uint8_t*)advPtr;
+static  ble_gap_adv_data_t d;
+static  uint8_t m_adv_data[BLE_GAP_ADV_SET_DATA_SIZE_MAX];
+memcpy(m_adv_data,advPtr,MIN(advLen,BLE_GAP_ADV_SET_DATA_SIZE_MAX));
+  d.adv_data.p_data = m_adv_data;
   d.adv_data.len = advLen;
   d.scan_rsp_data.p_data = m_enc_scan_response_data;
   d.scan_rsp_data.len = m_enc_scan_response_data_len;
```
however I guess there is maybe better way?  the
`JSV_GET_AS_CHAR_ARRAY(advPtr, advLen, advDataVar);` could read it into static array directly?

gfwilliams, 2021-04-22:
Wow, thanks! That's a really good one to know.

Actually, SD 6 always had an issue with setting the scan response (because you had to update scan response and advertising at the same time IIRC and there weren't global buffers) so having static advertising buffers like this would mean that was trivial to solve as well.

Also, if you're on SD 6 you're probably on something with enough RAM that it doesn't matter.

fanoush, 2021-04-27:
as for the static buffers the examples and documentation mention that any change while advertising requires second set of buffers and switch :-(

https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.s132.api.v6.0.0/structble__gap__adv__data__t.html?cp=4_7_3_5_3_0_14

> - To update advertising data while advertising, provide new buffers to sd_ble_gap_adv_set_configure

gfwilliams, 2022-06-08:
Wow, just saw this again. Are you still hitting this?

Interestingly Bangle.js 2 has been running with non-static buffers for ages on 6.0.0 now. Having said that, while scan response is non-static, I believe the way we now write advertising data into JsVars ensures that it is now a flat string of data, so the pointer returned by JSV_GET_AS_CHAR_ARRAY should effectively be static.

But since for Espruino 6.0.0+ generally means nRF52840, we could definitely afford to drop <64 bytes of RAM on 2 buffers for this.

fanoush, 2022-06-10:
Yes, it is still an issue, just tried with Q3/Bangle 2, flashed S140-6.1.0 (have S140 dfu packages here https://github.com/fanoush/ds-d6/tree/master/espruino/DFU/Magic3) and it is broken I can't find the watch properly advertising and mostly can't reconnect via WebIDE over remembered connection. It does work sometimes. I can also see the device via nrfconnect  but it has no name and advertising data is random.

It is feature of 6.1.0 and up, as described in that devzone link they call 6.1.0 compatible with 6.0.0 because the requirement is documented here https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.s132.api.v6.0.0/structble__gap__adv__data__t.html?cp=4_7_3_6_3_0_14 and worked by coincidence with 6.0.0 when not static.

BTW so far 6.0.0 was behaving better with espruino than 6.1.x with advertising fixed - @enaon reported connection issues from espruino device to other devices with 6.1.0 and 6.1.1 - opening connection seemed to be slower with 6.1.x.
Could be also because of new features in 6.1 `The main new feature for s140_nrf52_6.1.0 compared to s140_nrf52_6.0.0 is the full support for all mandatory LE Advertising Extensions features and qualified LE Coded PHY feature.` Maybe it needs some extra handling in espruino to work better. We'll see when you eventually move to 6.1 with Bangle 2

fanoush, 2023-05-08:
funny thing, I just hit one nasty bug with S140 6.1.1 - writing 4096 page of zeroes to internal flash crashes the device. The 6.0.0 works!
It is because of this issue https://devzone.nordicsemi.com/f/nordic-q-a/40088/sd_flash_write-cause-nrf_fault_id_sd_assert and is related to this line https://github.com/espruino/Espruino/blob/master/targets/nrf5x/jshardware.c#L2617 - value 4096 is not safe here with 6.1.x so beware when possibly upgrading. Reducing to 2048 as per comment https://devzone.nordicsemi.com/f/nordic-q-a/40088/sd_flash_write-cause-nrf_fault_id_sd_assert/244918 seems to work.

fanoush, 2023-05-09:
And BTW I was looking into this 6.1.x advertising issue again and made a bit more invasive changes. I noticed it is not just this one place I patched, there is also 2 more places https://github.com/search?q=repo%3Aespruino%2FEspruino%20sd_ble_gap_adv_set_configure&type=code one even in common jswrap_bluetooth.c

So I moved static arrays out of this method and used it also here https://github.com/espruino/Espruino/blob/d244795aca22d0787b349a8403171655953affe6/targets/nrf5x/bluetooth.c#L2621 and also I added symmetrical jsble_advertising_update_scanresponse to nrf52 source  below it and moved nrf5x code out of common code here https://github.com/espruino/Espruino/blob/d244795aca22d0787b349a8403171655953affe6/libs/bluetooth/jswrap_bluetooth.c#L1226 into this , do you want to have PR with this?

Also I am not sure why advertising is restarted around line here https://github.com/espruino/Espruino/blob/d244795aca22d0787b349a8403171655953affe6/libs/bluetooth/jswrap_bluetooth.c#L1226 and not here https://github.com/espruino/Espruino/blob/d244795aca22d0787b349a8403171655953affe6/targets/nrf5x/bluetooth.c#L2621 it is same type of code. I would remove the restart. Since both packets are now in static array and are known, both can be set, so it solves this comment https://github.com/espruino/Espruino/blob/d244795aca22d0787b349a8403171655953affe6/targets/nrf5x/bluetooth.c#L2620 if it was an issue and maybe it was the reason for the advertising restart too?

Also one more related thing - scan response is now also kept in variable here https://github.com/espruino/Espruino/blob/d244795aca22d0787b349a8403171655953affe6/libs/bluetooth/jswrap_bluetooth.c#L1200 maybe this can be removed when having static array? and funnily while this one is in JS variable (and advertising packet is not), there is actually no NRF.getScanResponse(Data?) to get it like there is https://www.espruino.com/Reference#l_NRF_getAdvertisingData . OTOH there is both  [NRF.setScanResponse(data)](https://www.espruino.com/Reference#l_NRF_setScanResponse) and [NRF.setAdvertising(data, options)](https://www.espruino.com/Reference#l_NRF_setAdvertising) so it is a bit of a mess (possibly for reasons).

fanoush, 2023-05-10:
ok made https://github.com/espruino/Espruino/pull/2367 but will retest with S132 3.0 and below, or maybe I can put the restart back into https://github.com/espruino/Espruino/pull/2367/commits/a6c41185c154880b7d8a4124b07e7469ffd0cf23#diff-19dab7787aab7c6471994842e692eceeb00f5a9d4742dbaa3374aae2d5051e5b around call to `err_code=jsble_advertising_update_scanresponse((uint8_t *)respPtr, respLen);`

PR #2367 (2023-05-26): for SD>5 use static buffers for advertising and scan response data
S140 6.1.x and up no longer makes its own copy of advertising data, provided buffers need to be valid all the time, also any change needs new set of buffers (otherwise it fails with NRF_ERROR_INVALID_STATE)

fixes #2000
closed 2023-05-26 by gfwilliams