espruino/Espruino #2013: StorageFile.write and ArrayBuffer

notEvil, 2021-06-03:
Hi,

`BufferArray` objects can be written directly with `Storage.write` but not with `StorageFile.write` (the latter performs something like `BufferArray.toString`). Is there another way to achieve this?

My (naive) impression is that if string objects of `BufferArray` objects were accessible, like `BufferArray` objects of `*Array` objects with `*Array.buffer`, the interface would be simpler (e.g. `Storage.write` wouldn't need a special case for `BufferArray`s).

notEvil, 2021-06-04:
Just found `E.toString(ArrayBuffer)` which is close enough, sorry for the distraction.
Still for `ArrayBuffer`s backed by normal strings it may still fail when there isn't enough memory left.

gfwilliams, 2021-06-07:
Hi - yes, you're right - this is definitely something that should be fixable pretty easily - we should be able to iterate over pretty much anything and write it to a file. I guess it might just have been because it was easier (as we ideally want to write a chunk of flash rather than one byte at a time) but it shouldn't be too difficult

notEvil, 2021-06-09:
Thanks!

Just fyi, I stopped using `StorageFile`, because of the "no 0xff" restriction, in favor of my own implementation (writes current size to first 2 bytes on flush) which better suits my use case.

gfwilliams, 2021-06-09:
Thanks - yes, I have been wondering about redoing StorageFile using packets of size+data. Obviously tweaking it would mean making it incompatible with data saved on older implementations, which will probably upset some people though

closed 2023-10-06 by gfwilliams