espruino/Espruino #2014: heatshrink.compress doesn't work with multibyte arrays

notEvil, 2021-06-09:
`hs=require('heatshrink'); hs.decompress(hs.compress(new Int16Array([256, 257, 258])))` returns `new Uint8Array([0, 1, 2]).buffer`. The documentation states "data - The data to decompress". Its easy to workaround though :)

gfwilliams, 2021-06-09:
Ahh - a lot of Espruino is like that. If you pass an array, it iterates over all the elements, treating each as a byte.

Imagine the case where you do just `hs.compress([256, 257, 258])` - there's no clue if it should be a byte array, int16, int32/etc

Doing this should work:

```
hs=require('heatshrink'); hs.decompress(hs.compress((new Int16Array([256, 257, 258])).buffer))
```

I think in this case the docs just need tweaking

notEvil, 2021-06-09:
Imo "treating each as a byte" shouldn't be. In most cases the size is known and constant (all string types, `*Array.byteLength / *Array.length`) and should be heeded. Functions which require a sequence of bytes shouldn't accept js lists.

I probably miss something here.

(`Storage.write` is also affected)

gfwilliams, 2022-02-01:
Just want to follow up on this - what do you expect:

```
hs=require('heatshrink'); 
r = hs.decompress(hs.compress(new Int16Array([256, 257, 258])))
```

to actually have in `r`?

notEvil, 2022-02-01:
`(new Int16Array([256, 257, 258])).buffer` because thats what I'd expect `hs.compress` to use instead of generic iteration with implicit reinterpret cast to uint8.

gfwilliams, 2022-02-01:
Ok, then I think this is something we just have to document.

`hs.compress` doesn't store any information about the type of data being given to it. It just compresses the raw data, so once it's compressed, it has no idea how to reconstruct it to anything other than bytes.

The available options are:

```
=new Uint8Array([0, 1, 2]).buffer
=new Uint8Array([0, 1, 1, 1, 2, 1]).buffer
```

And out of those, I think probably the first (existing) option is the most sane

notEvil, 2022-02-01:
I don't think the point is that `hs.compress` doesn't preserve the type it was given. It actually doesn't preserve the data. Imo `hs.compress` should always use the buffer when it gets an array.

gfwilliams, 2022-02-01:
So you think it should be option 2 that we get? `new Uint8Array([0, 1, 1, 1, 2, 1]).buffer`

notEvil, 2022-02-01:
I don't see any reasonable argument for option 1 (except historical consistency). Type information is lost anyway and preserving it is not heatshrinks responsibility. A compression algorithm is about preserving data, not meta data.

I think you can leave the code as is and update the documentation to remind the user of this behavior. Or change the behavior and leave the documentation as is ^^

commit 6a83202c (2023-08-16): Heatshrink docs, fix #2014
closed 2023-08-16 by gfwilliams