espruino/Espruino #2015: Is there a implantation of gethostbyname for Linux and  Wiznet?

MaBecker, 2021-06-14:
Sometime you only get hostnames and no ip...

Searching gethostbyname in repo Espruino get lots of implementation and wifi has a documented call in the reference documentation. 

How does Linux and builds including Wiznet ethernet  handle this?

gfwilliams, 2021-06-14:
On Linux I believe it just calls into the OS-provided gethostbyname function.

On WIZnet there's a special implementation just for WIZnet devices I believe, and I guess on ESPxx it calls into a function provided by Espressif?

MaBecker, 2021-06-14:
>  ESPxx it calls into a function provided by Espressif?

Yes there is a API call for this as part of the Wifi implementation.

> On Linux I believe it just calls into the OS-provided gethostbyname function.

> On Wiznet there's a special implementation just for WIZnet devices I believe

Yes, just check with `require("Ping").ping({ address: 'ahost' }, function(err, data) {console.log(data);});` 

Created an alias for  a web server on port 80, named it `ahost` and check with nslookup.

How to implement getHostByname for LINUX and Wiznet builds?

gfwilliams, 2021-06-14:
I don't understand - I thought it existed already internally? Or you want the function to be accessible from JS?

MaBecker, 2021-06-14:
> Or you want the function to be accessible from JS?

Yes, that's what I have in mind.

gfwilliams, 2021-06-15:
Ok, then it's easy. There's already an API for it in Espruino that should get implemented on most devices: https://github.com/espruino/Espruino/blob/master/libs/network/network.h#L82

So you just add a JS function and then call that. **However** if I recall there are hacks on some platforms since the current API is synchronous but on some systems (AT command, others?) it's async