espruino/Espruino #2017: Linux build seg faults on quit() if JsVars leaked

GaryOtt, 2021-06-17:
I have found a problem in Espruino while trying to debug a custom extension that's leaking JsVars. I hoped to find which variables they are by calling `quit()` but get a segmentation fault:

```
>[calls to code that leaks JsVars here]
>
>quit()
=undefined
>USED VAR #84:  #84[r0,l1] Segmentation fault (core dumped)
```

Looking at this code, `execInfo.root` is set to 0 by `jsiKill()` causing the seg fault in `jsvShowAllocated()`:
https://github.com/espruino/Espruino/blob/68a501e115bea86b4447e9a09723595109c3b93d/targets/linux/main.c#L506-L516

`jsiKill()` calls `jspKill()` which calls `jspSoftKill()` which contains the offending line:
https://github.com/espruino/Espruino/blob/68a501e115bea86b4447e9a09723595109c3b93d/src/jsparse.c#L2930

`jsvShowAllocated()` calls `jsvTrace()` which calls `_jsvTrace()` which calls `jsvCountJsVarsUsed()` at line 3669 which is where the segmentation fault happens:
https://github.com/espruino/Espruino/blob/68a501e115bea86b4447e9a09723595109c3b93d/src/jsvar.c#L3021

I tried changing the order of function calls in `main()` but unsurprisingly, it wasn't that simple.

gfwilliams, 2021-06-17:
Thanks - this sounds like a regression. I guess the easy option is to ensure jsvCountJsVarsUsed doesn't do the check if `execInfo.root = 0`?

GaryOtt, 2021-06-17:
I think this is what you meant:

```c
size_t jsvCountJsVarsUsed(JsVar *v) {
  // don't count 'root' when getting sizes
  if ((execInfo.root) && (v != execInfo.root)) execInfo.root->flags |= JSV_IS_RECURSING;
  // we do this so we don't count the same item twice, but don't use too much memory
  size_t c = _jsvCountJsVarsUsedRecursive(v, false);
  _jsvCountJsVarsUsedRecursive(v, true);
  // restore recurse flag
  if ((execInfo.root) && (v != execInfo.root)) execInfo.root->flags &= ~JSV_IS_RECURSING;
  return c;
}
```

It seems to fix the problem. I'll submit a pull request today.

gfwilliams, 2021-06-18:
Fixed with  #2018  - thanks!

closed 2021-06-18 by gfwilliams