espruino/Espruino #2019: Promise returned by then doesn't resolve parent

notEvil, 2021-06-27:
Hi,

this
```js
function f(promise) { promise.then(() => new Promise((resolve) => resolve())).then(() => console.log('here')); }

var a;

setTimeout(() => {
  print('works');
  a = Promise.resolve();
  f(a); // prints here
}, 0);

setTimeout(() => {
  print('doesn\'t');
  a = Promise.resolve();
  setTimeout(() => f(a), 0); // doesn't print here
}, 1000);
```
reproduces the issue. The relevant pieces are `new Promise` and the event loop iteration between `a = ...` and `f(a)`.
Maybe related to https://github.com/espruino/Espruino/issues/1476?

gfwilliams, 2021-06-28:
Thanks! That's an interesting one. 

Just tried and if you do:

```
function f(promise) { promise.then(() => { print("one");return new Promise((resolve) => resolve())}).then(() => console.log('here')); }
```

it prints `one` - so the first promise gets called ok - it just doesn't keep calling down the chain.

notEvil, 2021-06-28:
Found that
```
a = Promise.resolve()
a.then(print).then(print)
```
is sufficient (again with the second line in a different iteration) :)

notEvil, 2021-07-01:
I think I found the source of the issue in https://github.com/espruino/Espruino/blob/5c70d8ce4e907f6617620fbecc85e05d778f80af/src/jswrap_promise.c#L364.

When both lines are run in the same iteration, `Promise.resolve` queues a resolve and `.then` are added to an unresolved promise. On the other hand, when the second line is run in a different iteration, `.then` is called on a resolved promise, leading to the aforementioned line. It queues the callback but doesn't consider its return value nor resolve the chained promise (which will be created later in `.then`). The latter is no responsibility of `..._add` though.

gfwilliams, 2021-07-01:
Ahh - great! Thanks for digging around and figuring this out.

I'm just pushing a fix for it now.

closed 2021-07-01 by gfwilliams

notEvil, 2021-07-01:
Thanks for the fix!

commit e0490c0a (2021-07-01): Promise: Fix issue with .then().then(...) on an already-resolved promise (fix #2019)