espruino/Espruino #2020: Is atob broken, or is the way using it wrong?

MaBecker, 2021-06-28:
Running this sample code on Espruino for Linux (Mac OS X)

This is all about shrinking and converting JS to a compact and transferable way to be send over BLE_Uart.

Did not expect to have such many \0 in atob() result.

```js
>process.env
={
  VERSION: "2v09.11",
  GIT_COMMIT: "e533654b1",
  BOARD: "LINUX",
  FLASH: 262144, SPIFLASH: 262144, STORAGE: 262144, RAM: 0,
  SERIAL: "00000000-addeadde-addeadde",
  CONSOLE: "Telnet",
  MODULES: "Flash,Storage,hea" ... "crypto,tensorflow",
  EXPTR: 81838224 }
>
```

```js
var sampleJS = {
  "product": "Live JSON generator",
  "version": 3.1,
  "releaseDate": "2014-06-25T00:00:00.000Z",
  "demo": true,
  "person": {
    "id": 12345,
    "name": "John Doe",
    "phones": {
      "home": "800-123-4567",
      "mobile": "877-123-1234"
    },
    "email": [
      "jd@example.com",
      "jd@example.org"
    ],
    "dateOfBirth": "1980-01-02T00:00:00.000Z",
    "registered": true,
    "emergencyContacts": [
      {
        "name": "Jane Doe",
        "phone": "888-555-1212",
        "relationship": "spouse"
      },
      {
        "name": "Justin Doe",
        "phone": "877-123-1212",
        "relationship": "parent"
      }
    ]
  }
};

var s_sampleJS = E.toJS(sampleJS);
var cs_sampleJS = hs.compress(s_sampleJS);
var scs_sampleJS= E.toString(cs_sampleJS);
var bscs_sampleJS = atob(scs_sampleJS);

/*
dump()
var s_sampleJS = "{product:\"Live JSON generator\",version:3.1,releaseDate:\"2014-06-25T00:00:00.000Z\",demo:true,person:{id:12345,name:\"John Doe\",phones:{home:\"800-123-4567\",mobile:\"877-123-1234\"},email:[\"jd@example.com\",\"jd@example.org\"],dateOfBirth:\"1980-01-02T00:00:00.000Z\",registered:true,emergencyContacts:[{name:\"Jane Doe\",phone:\"888-555-1212\",relationship:\"spouse\"},{name:\"Justin Doe\",phone:\"877-123-1212\",relationship:\"parent\"}]}}";
var cs_sampleJS = new Uint8Array([189, 220, 46, 86, 251, 37, 214, 199, 116, 157, 72, 169, 150, 155, 181, 150, 65, 74, 169, 211, 233, 210, 11, 61, 150, 221, 101, 185, 88, 110, 150, 251, 148, 138, 88, 18, 6, 229, 115, 180, 219, 237, 211, 169, 156, 186, 99, 44, 185, 89, 109, 150, 91, 13, 206, 203, 68, 12, 131, 101, 22, 131, 50, 152, 76, 102, 146, 217, 132, 218, 91, 50, 154, 213, 38, 19, 9, 208, 8, 73, 112, 8, 25, 133, 104, 100, 13, 146, 203, 109, 183, 206, 174, 151, 43, 173, 150, 89, 112, 30, 4, 59, 10, 247, 105, 178, 78, 166, 51, 41, 156, 210, 107, 44, 183, 88, 109, 163, 160, 169, 86, 251, 69, 186, 65, 68, 183, 217, 69, 64, 220, 45, 1, 224, 108, 183, 49, 0, 32, 96, 66, 225, 206, 7, 192, 203, 68, 225, 75, 69, 0, 205, 166, 225, 176, 72, 128, 216, 173, 38, 240, 66, 161, 77, 230, 225, 81, 8, 33, 200, 175, 178, 194, 224, 54, 16, 176, 51, 170, 220, 138, 213, 100, 160, 89, 111, 4, 224, 110, 2, 32, 101, 214, 49, 232, 34, 224, 66, 5, 174, 1, 179, 200, 171, 168, 112, 84, 225, 83, 236, 212, 43, 77, 202, 233, 104, 34, 131, 49, 156, 151, 65, 87, 3, 49, 1, 3, 50, 87, 30, 213, 6, 207, 105, 185, 138, 65, 3, 2, 164, 5, 104, 33, 128, 16, 88, 54, 123, 45, 186, 199, 121, 161, 166, 193, 186, 88, 108, 119, 68, 240, 53, 187, 218, 220, 107, 10, 176, 21, 185, 73, 160, 128, 0, 203, 102, 160, 0, 83, 65, 0, 65, 36, 7, 108, 56, 131, 105, 12, 3, 115, 180, 90, 110, 1, 176, 110, 119, 11, 125, 214, 230, 42, 5, 96, 8, 240, 112, 104, 33, 192, 94, 197, 182, 75, 31, 35, 112, 176, 131, 193, 63, 2, 62, 6, 187, 125, 190, 128]).buffer;
var scs_sampleJS = "\xBD\xDC.V\xFB%\xD6\xC7t\x9DH\xA9\x96\x9B\xB5\x96AJ\xA9\xD3\xE9\xD2\v=\x96\xDDe\xB9Xn\x96\xFB\x94\x8AX\x12\6\xE5s\xB4\xDB\xED\xD3\xA9\x9C\xBAc,\xB9Ym\x96[\r\xCE\xCBD\f\x83e\x16\x832\x98Lf\x92\xD9\x84\xDA[2\x9A\xD5&\x13\t\xD0\bIp\b\x19\x85hd\r\x92\xCBm\xB7\xCE\xAE\x97+\xAD\x96Yp\x1E\4;\n\xF7i\xB2N\xA63)\x9C\xD2k,\xB7Xm\xA3\xA0\xA9V\xFBE\xBAAD\xB7\xD9E@\xDC-\1\xE0l\xB71\0 `B\xE1\xCE\7\xC0\xCBD\xE1KE\0\xCD\xA6\xE1\xB0H\x80\xD8\xAD&\xF0B\xA1M\xE6\xE1Q\b!\xC8\xAF\xB2\xC2\xE06\x10\xB03\xAA\xDC\x8A\xD5d\xA0Yo\4\xE0n\2 e\xD61\xE8\"\xE0B\5\xAE\1\xB3\xC8\xAB\xA8pT\xE1S\xEC\xD4+M\xCA\xE9h\"\x831\x9C\x97AW\31\1\32W\x1E\xD5\6\xCFi\xB9\x8AA\3\2\xA4\5h!\x80\x10X6{-\xBA\xC7y\xA1\xA6\xC1\xBAXlwD\xF05\xBB\xDA\xDCk\n\xB0\x15\xB9I\xA0\x80\0\xCBf\xA0\0SA\0A$\7l8\x83i\f\3s\xB4Zn\1\xB0nw\v}\xD6\xE6*\5`\b\xF0ph!\xC0^\xC5\xB6K\x1F#p\xB0\x83\xC1?\2>\6\xBB}\xBE\x80";
var bscs_sampleJS = "\0\0\x15\xB4\1\0\0\7^\0\5\0\v\0\7\1\x89\f\0\x1E\0\r-\0\06\0\0\b\x85\2\0\x0F\1\x8A\x88\3\x90\0\x17T\1\0\0\t\0\0(\0\0\7\0\x10\f\0\4\0\x0E\3\0\7b\x9C\0\x1E\3\0\2\x94H\0>0\0!\0\r\0\5\xD4\06\2\2\1~\0\02^\\\3\3\2\0\2\0\0\x1F\0\4\x80\0\x97\xC0\"\0\vf\x9F\0\0)\0\2\2\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0";
*/

```

MaBecker, 2021-06-28:
online encoding of scs_sampleJS

https://www.base64encode.net/share/beLaH

MaBecker, 2021-06-29:
Ups, mixed atob and btoa 

var ascs_sampleJS = btoa(scs_sampleJS);

gfwilliams, 2021-06-29:
So it works fine now?

MaBecker, 2021-06-29:
Yes and No ....  `E.toJS(sampleJS)` is smaller than  `ascs_sampleJS`, 

```js
var s_sampleJS = "{product:\"Live JSON generator\",version:3.1,releaseDate:\"2014-06-25T00:00:00.000Z\",demo:true,person:{id:12345,name:\"John Doe\",phones:{home:\"800-123-4567\",mobile:\"877-123-1234\"},email:[\"jd@example.com\",\"jd@example.org\"],dateOfBirth:\"1980-01-02T00:00:00.000Z\",registered:true,emergencyContacts:[{name:\"Jane Doe\",phone:\"888-555-1212\",relationship:\"spouse\"},{name:\"Justin Doe\",phone:\"877-123-1212\",relationship:\"parent\"}]}}";
var cs_sampleJS = new Uint8Array([189, 220, 46, 86, 251, 37, 214, 199, 116, 157, 72, 169, 150, 155, 181, 150, 65, 74, 169, 211, 233, 210, 11, 61, 150, 221, 101, 185, 88, 110, 150, 251, 148, 138, 88, 18, 6, 229, 115, 180, 219, 237, 211, 169, 156, 186, 99, 44, 185, 89, 109, 150, 91, 13, 206, 203, 68, 12, 131, 101, 22, 131, 50, 152, 76, 102, 146, 217, 132, 218, 91, 50, 154, 213, 38, 19, 9, 208, 8, 73, 112, 8, 25, 133, 104, 100, 13, 146, 203, 109, 183, 206, 174, 151, 43, 173, 150, 89, 112, 30, 4, 59, 10, 247, 105, 178, 78, 166, 51, 41, 156, 210, 107, 44, 183, 88, 109, 163, 160, 169, 86, 251, 69, 186, 65, 68, 183, 217, 69, 64, 220, 45, 1, 224, 108, 183, 49, 0, 32, 96, 66, 225, 206, 7, 192, 203, 68, 225, 75, 69, 0, 205, 166, 225, 176, 72, 128, 216, 173, 38, 240, 66, 161, 77, 230, 225, 81, 8, 33, 200, 175, 178, 194, 224, 54, 16, 176, 51, 170, 220, 138, 213, 100, 160, 89, 111, 4, 224, 110, 2, 32, 101, 214, 49, 232, 34, 224, 66, 5, 174, 1, 179, 200, 171, 168, 112, 84, 225, 83, 236, 212, 43, 77, 202, 233, 104, 34, 131, 49, 156, 151, 65, 87, 3, 49, 1, 3, 50, 87, 30, 213, 6, 207, 105, 185, 138, 65, 3, 2, 164, 5, 104, 33, 128, 16, 88, 54, 123, 45, 186, 199, 121, 161, 166, 193, 186, 88, 108, 119, 68, 240, 53, 187, 218, 220, 107, 10, 176, 21, 185, 73, 160, 128, 0, 203, 102, 160, 0, 83, 65, 0, 65, 36, 7, 108, 56, 131, 105, 12, 3, 115, 180, 90, 110, 1, 176, 110, 119, 11, 125, 214, 230, 42, 5, 96, 8, 240, 112, 104, 33, 192, 94, 197, 182, 75, 31, 35, 112, 176, 131, 193, 63, 2, 62, 6, 187, 125, 190, 128]).buffer;
var scs_sampleJS = "\xBD\xDC.V\xFB%\xD6\xC7t\x9DH\xA9\x96\x9B\xB5\x96AJ\xA9\xD3\xE9\xD2\v=\x96\xDDe\xB9Xn\x96\xFB\x94\x8AX\x12\6\xE5s\xB4\xDB\xED\xD3\xA9\x9C\xBAc,\xB9Ym\x96[\r\xCE\xCBD\f\x83e\x16\x832\x98Lf\x92\xD9\x84\xDA[2\x9A\xD5&\x13\t\xD0\bIp\b\x19\x85hd\r\x92\xCBm\xB7\xCE\xAE\x97+\xAD\x96Yp\x1E\4;\n\xF7i\xB2N\xA63)\x9C\xD2k,\xB7Xm\xA3\xA0\xA9V\xFBE\xBAAD\xB7\xD9E@\xDC-\1\xE0l\xB71\0 `B\xE1\xCE\7\xC0\xCBD\xE1KE\0\xCD\xA6\xE1\xB0H\x80\xD8\xAD&\xF0B\xA1M\xE6\xE1Q\b!\xC8\xAF\xB2\xC2\xE06\x10\xB03\xAA\xDC\x8A\xD5d\xA0Yo\4\xE0n\2 e\xD61\xE8\"\xE0B\5\xAE\1\xB3\xC8\xAB\xA8pT\xE1S\xEC\xD4+M\xCA\xE9h\"\x831\x9C\x97AW\31\1\32W\x1E\xD5\6\xCFi\xB9\x8AA\3\2\xA4\5h!\x80\x10X6{-\xBA\xC7y\xA1\xA6\xC1\xBAXlwD\xF05\xBB\xDA\xDCk\n\xB0\x15\xB9I\xA0\x80\0\xCBf\xA0\0SA\0A$\7l8\x83i\f\3s\xB4Zn\1\xB0nw\v}\xD6\xE6*\5`\b\xF0ph!\xC0^\xC5\xB6K\x1F#p\xB0\x83\xC1?\2>\6\xBB}\xBE\x80";
var bscs_sampleJS = "vdwuVvsl1sd0nUiplpu1lkFKqdPp0gs9lt1luVhulvuUilgSBuVztNvt06mcumMsuVltllsNzstEDINlFoMymExmktmE2lsymtUmEwnQCElwCBmFaGQNksttt86ulyutlllwHgQ7Cvdpsk6mMymc0msst1hto6CpVvtFukFEt9lFQNwtAeBstzEAIGBC4c4HwMtE4UtFAM2m4bBIgNitJvBCoU3m4VEIIcivssLgNhCwM6rcitVkoFlvBOBuAiBl1jHoIuBCBa4Bs8irqHBU4VPs1CtNyuloIoMxnJdBVwMxAQMyVx7VBs9puYpBAwKkBWghgBBYNnstusd5oabBulhsd0TwNbva3GsKsBW5SaCAAMtmoABTQQBBJAdsOINpDANztFpuAbBudwt91uYqBWAI8HBoIcBexbZLHyNwsIPBPwI+Brt9voA=";

```

Is there another possibility to get this JSON5 smaller? This is all about reducing bytes to be send.

gfwilliams, 2021-06-29:
Do you mean `bscs_sampleJS`?

Well, there are two answers...

* Can you send binary data over the BLE UART? If so, `cs_sampleJS`/bscs_sampleJS` is your best option (they are effectively the same data)
* If you need to just send ASCII and the JSON you send is primarily ASCII bytes then yes, I think `E.toJS` is your best bet

As for sending binary, Espruino itself transmits XON/XOFF characters when it is a peripheral when its input buffers get full. It doesn't interpret XON/XOFF from other devices unless you explicitly enable it on Serial (IIRC).  Chances are, you'll be fine sending binary data over BLE UART, however you will still have to come up with a way of 'framing it'

closed 2021-06-29 by gfwilliams