espruino/Espruino #2021: The new Espruino drawLineAA bugs

tyrone1984, 2021-07-01:
I  built and tested the new Espruino, and the drawLineAA seems had bugs to dotted line.
 Is  It a bug?  Or just me  had this problem?

gfwilliams, 2021-07-02:
What command are you using? Did it work on earlier firmware? And do you see it on the emulator?

If you try a command line `g.drawLineAA(0,0,200,240)` in the emulator you'll see a slight 'beating':

[image]

This is to be expected as when the line converges on a single pixel, that one pixel is brighter than two half-brightness pixels. The fix would really be to adjust the gamma curve on the LCD to ensure that 50% brightness was exactly the same as 1 pixel on, 1 pixel off

tyrone1984, 2021-07-03:
Thanks,
I found the same problem after  tested  several earlier built, so i adjusted the gamma curve,  But it's really a hard work, :-( so i droped

gfwilliams, 2021-07-05:
So are you saying this is an issue that only recently occurred? As far as I know it would have been present in `drawLineAA` since it was created.

If there's an easy way to do gamma correction when working out how to blend pixels then I'm all for pulling it in though.

gfwilliams, 2021-09-14:
any thoughts on the above?

closed 2022-02-01 by gfwilliams