espruino/Espruino #2022: ESP32 boot failed

vidalouiswang, 2021-07-07:
Espruino Version: 2v09
commit: 30a9f3824ccdae3de72394691a5e8f9162ec3da6

The following information is from serial:

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:2668
load:0x40078000,len:7304
load:0x40080000,len:5312
entry 0x40080274
E (548) spiram: SPI RAM enabled but initialization failed. Bailing out.

[JASSERT(length>0) FAILED AT src/jsvar.c:4199
  #1[r1,l2] Object { 
    #2[r1,l2] Name String [1 blocks] "\xFF"      #3[r1,l1] Object { 
        #6[r1,l2] Name String [2 blocks] "timers"          #8[r2,l0] Array(0) [ ] 
        #9[r1,l2] Name String [2 blocks] "watches"          #11[r2,l0] Array(0) [ ] 
        #14[r1,l2] Name String [1 blocks] "net"          #12[r1,l1] String [2 blocks] "\5\0\0\0\0\xFF\xFF\xFF\0\0\0\0"  
      } 
  }
HALTING.
`

vidalouiswang, 2021-07-07:
After remove bluetooth support from ESP32.py, it boot successfully.

MaBecker, 2021-11-09:
looks like the variables in board file are to high.

Ritika-Das, 2022-01-28:
Any updates on this? Had to remove Bluetooth support, then it worked for the time being.

Ritika-Das, 2022-01-28:
I am getting this error after running `sudo minicom --baudrate 115200 --device /dev/ttyUSB0`
```
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9f1c: 16f95ea5 ffffffff ffffffff
Core 0 register dump:
PC      : 0x400e9f22  PS      : 0x00060730  A0      : 0x800eb96f  A1      : 0x3ffd6870  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00001800  A5      : 0x7fffffff  
A6      : 0x00000000  A7      : 0x3ffc745c  A8      : 0x800e9f1f  A9      : 0x3ffd6850  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x000003ea  
A14     : 0x000003f9  A15     : 0x00000000  SAR     : 0x00000016  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x40095d34  LEND    : 0x40095d3f  LCOUNT  : 0x00000000  

Backtrace: 0x400e9f22:0x3ffd6870 0x400eb96c:0x3ffd68a0 0x400fa87d:0x3ffd68c0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:2668
load:0x40078000,len:7304
load:0x40080000,len:5312
entry 0x40080274
E (429) spiram: SPI RAM enabled but initialization failed. Bailing out.
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:2668
load:0x40078000,len:7304
load:0x40080000,len:5312
entry 0x40080274
E (429) spiram: SPI RAM enabled but initialization failed. Bailing out.

 ____                 _ 
|  __|___ ___ ___ _ _|_|___ ___ 
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v11.61 (c) 2021 G.Williams

Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
```

gfwilliams, 2024-02-02:
I'm pretty sure this is fixed now

closed 2024-02-02 by gfwilliams