espruino/Espruino #2023: Cannot "turn off" Banglejs after flashing 2v09

bernhardreiter, 2021-07-11:
Having flashed  a watch from 2v08 to 2v09 (following the instructions and shown in the about infos), now I cannot turn the watch off.

* Selecting "Settings", 
*  selecting "turn off" from the last but one entry.
* press BTN2

Observation: The clock does a "reset",
it says "sw reset" and "DFU ZY08" and is back to the main sceen with the clock.

Expectation: screen stays black, just is did before flashing

Side observation: subjectively I'd say that buttons were a little less responsive after flashing.

gfwilliams, 2021-07-11:
Thanks - I'll look into this. I'm on holiday the next 2 weeks though and if you need to turn off you should be able to do so through the bootloader (hold btn1+2 for around 10 seconds). It's possible it is already fixed in 'cutting edge' builds though.

bernhardreiter, 2021-07-12:
@gfwilliams thanks for the hint and have nice holidays.

kuleszdl, 2021-07-31:
@bernhardreiter Did you reset something after updating to 2v09? For me, turning off used to work in 2v09, however, I resetted all apps and settings after updating the firmware.

bernhardreiter, 2021-08-01:
> Did you reset something after updating to 2v09? 

No. (Should I have done so? Do not remember this from the flashing instructions.)

gfwilliams, 2021-08-02:
There was actually a firmware change, so 'turn off' did `Bangle.softOff` and not `Bangle.off`.

If you also updated everything using the app loader, it's possible that `Bangle.softOff` actually works fine on 2v09 so everything would be ok.

bernhardreiter, 2021-08-02:
As far as I remember I did update everything I could with the app loader and afterwards had the problem. I'll recheck next time.

gfwilliams, 2021-08-13:
I can't reproduce this with new firmwares - they seem to work fine - so closing with 2v10 release.

closed 2021-08-13 by gfwilliams