espruino/Espruino #2034: [BangleJS] Leaking memory / garbage while calling Bangle.beep()

peeweek, 2021-07-30:
Hello, I'm currently on 2v09 and I figured out while working that my bangle memory was filling up slowly every time I was calling a `Bangle.beep()`. I have been using it naively so far when making interactions and I stumbled upon out of mem issues after a while. After debugging I ended up figuring out that every beep was consuming roughly 1 block of memory each call.

Is there a way for me to prevent that? Am I using it wrong?

I've osolated part of my code and made a sample logger with the free memory :  
```js
class beep {
  static select () {
    Bangle.beep(20,880).then(()=>Bangle.beep(50,1200));
  }
  static move() {
    Bangle.beep(20,440);
  }
  static deny() {
    Bangle.beep(20,440).then(()=>Bangle.beep(50,220));
  }
}

function logMem() {
  print(process.memory().free);
}

function onB1() {
  beep.select();
  logMem();
}
function onB2() {
  beep.move();
  logMem();
}
function onB3() {
  beep.deny();
  logMem();
}

// Handle Button Press
setWatch(onB1, BTN1, true);
setWatch(onB2, BTN2, true);
setWatch(onB3, BTN3, true);
```

[image]

gfwilliams, 2021-07-30:
This sounds bad - thanks for spotting it! I'll look into it first thing Monday

gfwilliams, 2021-08-02:
Looks like this was already fixed as part of https://github.com/espruino/Espruino/commit/221792142d8c6bf18a61d19b778e13a575a46e0f - it'll be in 2v10 (or cutting edge builds)

closed 2021-08-02 by gfwilliams