espruino/Espruino #2041: Unexpected execution of else statement

GaryOtt, 2021-08-29:
I encountered an error running uglified code that is demonstrated by the following one liner:

```JavaScript
if (true) { console.log("This should be the only visible output"); } else ((a) => { console.log("This IIFE should not run."); })(1)
```

I am getting the following unexpected output:

```
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v10 (c) 2021 G.Williams
>if (true) { console.log("This should be the only visible output"); } else ((a) => { console.log("This IIFE should not run."); })(1)
This should be the only visible output
This IIFE should not run.
Uncaught SyntaxError: Got ')' expected EOF
 at line 1 col 80
...ible output"); } else ((a) => { console.log("This IIFE shoul...
                              ^
 at line 1 col 128
...s IIFE should not run."); })(1)
                              ^
> 
```

I have replicated this issue on a Bangle.js running 2.10 and on latest builds for Linux and the ESP32.

gfwilliams, 2021-08-31:
Thanks! Pretty busy this week but I'll try and get something done here.

Just checked and: 

```
// same issue
if (0) ((a) => { console.log("This IIFE should not run."); })()
// fixes it
if (0) {((a) => { console.log("This IIFE should not run."); })()}
```

So it seems like it thinks there's a syntax error in the first instance, but even so it still continues execution

GaryOtt, 2021-08-31:
Ah! I can't believe I didn't think to try it as the statement of an `if`!

gfwilliams, 2021-08-31:
Added a minor tweak which solves the execution, but we still have a parsing issue where the parser doesn't like `if (0) ((a) => { console.log("This IIFE should not run."); })()`

gfwilliams, 2021-09-17:
Just to clear up the remaining issue:

```
// ok
if (1) (a => console.log("A test") )()
// ok
if (1) ((a) => console.log("A test") )()
// ok
if (0) (a => console.log("A test") )() 

// NOT OK
if (0) ((a) => console.log("A test") )()  

// Uncaught SyntaxError: Got => expected ',' at line 1 col 13
//if (0) ((a) => console.log("A test") )()
//            ^
```

Even `if (0) ((a) => 1 )()` fails.

The issue is at:

https://github.com/espruino/Espruino/blob/5d22d35f77a18d09b1bb33e93920c99393222029/src/jsparse.c#L1461-L1462

Basically when not executing, `jspeAssignmentExpression` parses everything, but returns 0 - but it needs to return a name so we can detect an arrow function.

Putting fix in now

gfwilliams, 2021-09-17:
Ok, the diff for this is:

```
diff --git a/src/jsparse.c b/src/jsparse.c
index 586fd810f..e4d3788e1 100644
--- a/src/jsparse.c
+++ b/src/jsparse.c
@@ -855,27 +855,33 @@ NO_INLINE JsVar *jspeFunctionCall(JsVar *function, JsVar *functionName, JsVar *t
 // Find a variable (or built-in function) based on the current scopes
 JsVar *jspGetNamedVariable(const char *tokenName) {
   JsVar *a = JSP_SHOULD_EXECUTE ? jspeiFindInScopes(tokenName) : 0;
-  if (JSP_SHOULD_EXECUTE && !a) {
-    /* Special case! We haven't found the variable, so check out
-     * and see if it's one of our builtins...  */
-    if (jswIsBuiltInObject(tokenName)) {
-      // Check if we have a built-in function for it
-      // OPT: Could we instead have jswIsBuiltInObjectWithoutConstructor?
-      JsVar *obj = jswFindBuiltInFunction(0, tokenName);
-      // If not, make one
-      if (!obj)
-        obj = jspNewBuiltin(tokenName);
-      if (obj) { // not out of memory
-        a = jsvAddNamedChild(execInfo.root, obj, tokenName);
-        jsvUnLock(obj);
-      }
-    } else {
-      a = jswFindBuiltInFunction(0, tokenName);
-      if (!a) {
-        /* Variable doesn't exist! JavaScript says we should create it
-         * (we won't add it here. This is done in the assignment operator)*/
-        a = jsvMakeIntoVariableName(jsvNewFromString(tokenName), 0);
+  if (!a) {
+    if (JSP_SHOULD_EXECUTE) {
+      /* Special case! We haven't found the variable, so check out
+       * and see if it's one of our builtins...  */
+      if (jswIsBuiltInObject(tokenName)) {
+        // Check if we have a built-in function for it
+        // OPT: Could we instead have jswIsBuiltInObjectWithoutConstructor?
+        JsVar *obj = jswFindBuiltInFunction(0, tokenName);
+        // If not, make one
+        if (!obj)
+          obj = jspNewBuiltin(tokenName);
+        if (obj) { // not out of memory
+          a = jsvAddNamedChild(execInfo.root, obj, tokenName);
+          jsvUnLock(obj);
+        }
+      } else {
+        a = jswFindBuiltInFunction(0, tokenName);
+        if (!a) {
+          /* Variable doesn't exist! JavaScript says we should create it
+           * (we won't add it here. This is done in the assignment operator)*/
+          a = jsvMakeIntoVariableName(jsvNewFromString(tokenName), 0);
+        }
       }
+    } else { // else we're not executing.
+      // But we must make a variable name anyway, so that we can detect
+      // arrow functions. https://github.com/espruino/Espruino/issues/2041
+      a = jsvNewWithFlags(JSV_NAME_STRING_0);
     }
   }
   return a;
```

but actually it breaks a bunch of other stuff so I need to look into why more closely

GaryOtt, 2021-10-05:
Trying this diff for myself had similar results. I don't know if the below yields any clues but I thought I'd share on the off chance.

```
Uncaught ReferenceError: "" is not defined
 at line 75 col 45
...   case "FILE_DEL": mode = MODE_DELETE;      break;
                              ^
in function called from system
```

I have been trying to understand the parser source code with the intent to help but it has taken more time than I've actually had to spare so far.

gfwilliams, 2022-02-01:
This is fixed now