espruino/Espruino #2043: Bangle.js showMenu customisation

gfwilliams, 2021-08-31:
http://forum.espruino.com/conversations/367303/

The original graphical_menu module allows font size to be changed, but in Bangle.js we hard-code this and also don't call predraw - we should probably fix this so menus can be customised

mbattistini, 2021-08-31:
It would be great if it were uniform with the fonts also used in the app selection, just like for other similar systems (Smartwatch), it would make it more accessible, like for people like me who need glasses, and consolidated (less chaotic and confusing).

gfwilliams, 2021-08-31:
I'm not sure what you mean? You mean to use the fonts that other apps use?

It actually does, but it's just that the apps are a complete free for all, so they use whatever they fancy. If the apps put up a menu or message then they will all use the same font

mbattistini, 2021-08-31:
No, obviously each app is separate BUT having only one font, readable and maybe selectable (even just for the size), uniform for the launcher, at least the default one, and for the sub-menus of the settings page, I think it would create less confusion.

gfwilliams, 2021-12-15:
`predraw` has since been added, but you can also customise by replacing showMenu now

closed 2021-12-15 by gfwilliams