espruino/Espruino #2044: Graphics bitmap font size in X and Y

gfwilliams, 2021-09-02:
Currently we allow bitmap fonts to be scaled up by a factor of 2/3/4/etc

We *could* allow them to be scaled up by a different factor in each dimension, without too many painful changes.

Is this something that would be useful? I can imagine that maybe in menus on Bangle.js 1 it might be better to triple the height but only double the width...

gfwilliams, 2021-09-20:
Just a note to say I made some initial tweaks and it doesn't look too bad.

The only issue is how we set the font up... I guess we could just extend `setFont` to `setFont("FontName",x,y)` although I recently added `g.setFont("FontName:size")` which is nice in that the font is just one text string, which is easily changed. 

So we could just tweak it to allow `g.setFont("FontName:XxY")` like `g.setFont("6x8:1x2")`

What does everyone think?

MaBecker, 2021-09-20:
my choice is g.setFont("6x8:1x2")

MaBecker, 2021-09-20:
Could this be available for vector font too?

gfwilliams, 2021-09-20:
Yes, it works for that too. Only gotcha is it's max of 63x63 pixel size when you scale separately because we've had to pack all the font info into 16 bits.

Still, it's pretty handy.

commit ad961bdf (2021-09-20): Allow setFont("FontName:1x2") for scaling in 2 directions (fix #2044)
closed 2021-09-20 by gfwilliams