espruino/Espruino #2046: BLE device no longer reachable

MaBecker, 2021-09-13:
From time to time a `WARNING: SD ERR 0x11 (BUSY) (bluetooth.c:1791)` occurrs, but code is without `NRF.setSecurity()`. Only a power cycle helps to  get that device connectable again. This occurs after a NUS connection stress test - connect every 120 sec from  gateway MQTT2BLE to a BLE device. Both devices run the latest firmware. 

https://github.com/espruino/Espruino/blob/0c9a8ee0f81804d184f41c14ba67b7c202514537/targets/nrf5x/bluetooth.c#L1784-L1797

gfwilliams, 2021-09-13:
So just to be sure I understand:

* Device `A` is being connected to by device `B`
* At some point, `A` cannot be connected to and keeps giving `WARNING: SD ERR 0x11 (BUSY) (bluetooth.c:1791)`
* The only way to connect again is to restart `A`. Restarting `B` has no effect

And roughly how long does this take to reproduce? Is it pretty obviously after the same amount of time each time?

Recently I've had issues with getting BUSY event due to MTU negotiation. I guess you could try compiling firmwares for both devices without the increased MTU and see if that fixes it?

MaBecker, 2021-09-13:
> * Device A is being connected to by device B

 Yes

> * At some point, A cannot be connected to and keeps giving WARNING: SD ERR 0x11 (BUSY) (bluetooth.c:1791)

   `A` say `No device found` when searching for `B`, so `A`cannot connect any more to `B`, same for the WebIDE, device is no longer visible.

> * The only way to connect again is to restart A. Restarting B has no effect

restart `B` so `A` can connect again to `B`
`B` gives the `WARNING: SD ERR 0x11 (BUSY) (bluetooth.c:1791)` when connecting  via WebIDE after the restart.

> Recently I've had issues with getting BUSY event due to MTU negotiation. I guess you could try compiling firmwares for both devices without the increased MTU and see if that fixes it?

Good point, will do.

gfwilliams, 2021-09-13:
How long does this take to reproduce? Is it pretty obviously after the same amount of time each time?

> restart B so A can connect again to B

Didn't we just say that `B` was the one connecting to `A`?

Or does restarting `A` have no effect - so it's to do with `B`'s Central connection to A? If `A` can no longer be found, chances are `B` is actually still connected to it and has got out of sync

gfwilliams, 2021-09-13:
Also, please could you try and get me the most minimal bit of code that reproduces this that is possible? Even better if device `A` is an entirely standard Espruino device not running any special code.

MaBecker, 2021-09-13:
> How long does this take to reproduce? Is it pretty obviously after the same amount of time each time?

last time it took 79 calls until device `B` was frozen - no longer visible or connectable via WebIDE 

* `A` and `B` are NRF52840 devices running on Espruino 2v10.31
* `A` is the gateway receiving mqtt messages and forward them via BLE-NUS to device `B`
* `B` is receiving data via BLE-NUS from device `A`

Minimal code snippet running on device `A`:

```js
var id = "ed:4f:05:32:73:ad random";
var dev;
var cmd = JSON.stringify({ a: "a", b: "b" });
var n = 0;
var e = 0;

bleSendJS = function() {
    NRF.requestDevice({ timeout: 5E3, filters: [{ id: id }] }).then(function(device) {
        print('r');
        dev = device;
        return require("ble_uart").connect(device);
    }).then(function(uart) {
        print('c');
        var data;
        uart.on('data', function(d) {
            data += d;
        });
        uart.write(`\x03\x10print(${cmd})\n`).then(function() {
            print("n:", ++n);
        });
        setTimeout(function() {
            uart.disconnect();
            print("got:", data);
        }, 2000);
    }).catch(function(err) {
        print("e:", ++e);
        print('e:', err);
    });
};
bleSendJS();

setInterval('bleSendJS();', 15E3);
```

output:
```
r
c
n: 1
got: undefined<- Serial1
>{
  "a": "a",
  "b": "b"
 }
>
```

Code is now running.

MaBecker, 2021-09-13:
With this snippet those errors occur multiple times on device `A`:

```
r
e: 1
e: Disconnected
Uncaught InternalError: BLE task completed that wasn't scheduled (SERVICE/NONE)
```

error is reproduce able with this snippet, took 105 calls :

```
r
c
n: 103
got: undefined<- Serial1
>{
  "a": "a",
  "b": "b"
 }
>
r
e: 3
e: Disconnected
e: 4
e: No device found matching filters
e: 5
e: No device found matching filters
e: 6
e: No device found matching filters
e: 7
e: No device found matching filters
e: 8
e: No device found matching filters
....
e: 18
e: No device found matching filters
e: 19
e: No device found matching filters
e: 20
e: No device found matching filters
e: 21
e: No device found matching filters
```

MaBecker, 2021-09-13:
Now both devices use the same MTU size of 131.

gfwilliams, 2021-09-13:
> Now both devices use the same MTU size of 131

try without any extension at all - so the default of 23. Even if both have 131 they still have to negotiate which could cause in issue. It may not be a long term solution but it would help narrow it down

MaBecker, 2021-09-14:
> Now both devices use the same MTU size of 131.

still running,  device `B` is still reachable, got 2500 calls and 142 errors of type:

```
e: Disconnected
Uncaught InternalError: BLE task completed that wasn't scheduled (SERVICE/NONE)
```

Is this MTU negotiation related too?

MaBecker, 2021-09-14:
Next run both devices with default MTU size 23

gfwilliams, 2021-09-14:
Am I right in thinking that device B could be a completely standard Espruino not running any code at all?

> still running, device B is still reachable

Ok, so what's the difference? I thought both devices would have had the same MTU size anyway?

> BLE task completed that wasn't scheduled (SERVICE/NONE)

I believe this is when getPrimaryServices completes, but Espruino doesn't think it was waiting for a response. What's happened is because you don't do any checking and just call bleSendJS every 15 seconds, in some cases bleSendJS has taken more than 15 seconds to connect to the device. When that happens you're then attempting to connect a second time while the first getPrimaryServices is in progress.

So you could add a check to ensure you're not trying to connect when the first connection is still in progress?

Although realistically there should be some checking in Espruino that should have errored already. I guess if we try doing two connections at the same time, it might be a way to reproduce the problem sooner and more reliably.

For now, to actually get your system working reliably I guess you could detect multiple non-connects and just call `NRF.reboot()`?

MaBecker, 2021-09-14:
> Am I right in thinking that device B could be a completely standard Espruino not running any code at all?

Yes, just a BLE device flashed with Espruino will do.

> Ok, so what's the difference? I thought both devices would have had the same MTU size anyway?

Well, I though the MTU was the same, but `A`was using MTU 131 and `B` was using 53.

> So you could add a check to ensure you're not trying to connect when the first connection is still in progress?

Yes this is all part of the more complex code that's used. Just the striped code does not contain those.

> For now, to actually get your system working reliably I guess you could detect multiple non-connects and just call NRF.reboot()?

On the `B` device? 

Will stick to MTU 131

MaBecker, 2021-09-14:
Latest test snippet with write to log and a catch for that reject.
 
```js
var f = require("Storage").open("log","a");
var id = "f0:62:84:f4:d3:a6 random";
var dev;
var cmd = JSON.stringify({ a: "a", b: "b" });
var nc = 0;
var ne = 0;

bleSendJS = function() {
    print('- - - - - -');
    NRF.requestDevice({ timeout: 5E3, filters: [{ id: id }] }).then(function(device) {
        print('r');
        dev = device;
        return require("ble_uart").connect(device);
    }).then(function(uart) {
        print('c');
        var data;
        uart.on('data', function(d) {
            data += d;
        });
        uart.write(`\x03\x10print(${cmd})\n`).then(function() {
            print("nc:",++nc);
            setTimeout(function() {
              uart.disconnect();
              print("got:",data);
            }, 2000);
        }).catch(function () {
            console.log("rejected - disconnect");
            uart.diconnect();
        });
    }).catch(function(err) {
        print("ne:",++ne);
        print('err:', err);
        f.write(getTime() + " " + nc + " " + ne + " " + err + "\n");
    });
};

NRF.setTxPower(8);

bleSendJS();

setInterval('bleSendJS();',2E4);
```

MaBecker, 2021-09-15:
> For now, to actually get your system working reliably I guess you could detect multiple non-connects and just call NRF.reboot()?

Ok, I guess you mean something like this:

```js
    NRF.on('disconnect', function(){
      setTimeout(()=>{NRF.restart();NRF.setTxPower(8);},500);
    });
```

gfwilliams, 2021-09-15:
Yes, although...

* After NRF.restart the system totally reboots, so setTxPower has no effect
* You'd need to detect multiple issues connecting (probably via promise rejection). 'disconnect' will get called for every disconnect

MaBecker, 2021-09-15:
Hmm, tested this snippet on a PuckJS with the WebIDE and there is no device restart, just a red light for a connect and a green for disconnect.

```js
NRF.on('connect', function(addr) {
  LED1.toggle();
  setTimeout(()=>{LED1.toggle();},250);
});

NRF.on('disconnect', function(addr) {
  LED2.toggle();
  setTimeout(()=>{LED2.toggle(); NRF.restart();},250);
});

```

Or do you mean the Nordic softdevice is restarted?

gfwilliams, 2021-09-15:
Oh, sorry - I actually meant E.reboot, but if NRF.restart does it (which it should) then that'd be preferable i guess

MaBecker, 2021-09-17:
Ups, closed by mistake....

MaBecker, 2021-09-17:
So the only reliable solution is to do a `E.reboot()` on device `B` after  the disconnect.

Is there any thing I could add and run over the weekend  to narrow this down?

MaBecker, 2021-09-22:
Any hints how to investigate?

Because device A is affected as well, it is just a question of time/calls.

gfwilliams, 2021-09-23:
So this even happens when you have the standard, non-expanded 23 byte MTU?

I guess you could try and add some logging to the BLE security-related events on your build. It is possible with some fiddling to just turn on Nordic's internal logging to UART which may help you.

MaBecker, 2021-09-23:
> It is possible with some fiddling to just turn on Nordic's internal logging to UART which may help you.

Wow, no clue, hope you can advise

gfwilliams, 2021-09-23:
You can try adding:

```
   'makefile' : [
     'DEFINES += -DNRF_LOG_ENABLED=1 -DNRF_LOG_FILTERS_ENABLED=0',
```

To BOARD.py, and possibly you need to set some default UART0 pins in targetlibs/nrf_sdk15/nrf_config/sdk_config.h

Sorry I can't be much help, but as you might guess I'm a bit busy with the KickStarter at the moment!

MaBecker, 2021-09-29:
Ok, using two PuckJS using SDK12

```make
'makefile' : [
    'DEFINES +=-DNRF_LOG_ENABLED=1 -DNRF_LOG_DEFAULT_LEVEL=4'
]
```

NRF_LOG_ENABLED=1 cause some errors 

```c
targets/nrf5x/bluetooth.c:896:6: error: conflicting types for 'nrf_log_frontend_std_0'
 void nrf_log_frontend_std_0(uint32_t severity_mid, char const * const p_str) {
      ^~~~~~~~~~~~~~~~~~~~~~
In file included from /Volumes/Espruino/repos/Espruino/targetlibs/nrf5x_12/components/libraries/log/nrf_log.h:87,
                 from targets/nrf5x/bluetooth.c:46:
/Volumes/Espruino/repos/Espruino/targetlibs/nrf5x_12/components/libraries/log/src/nrf_log_internal.h:206:6: note: previous declaration of 'nrf_log_frontend_std_0' was here
 void nrf_log_frontend_std_0(uint8_t severity, char const * const p_str);
      ^~~~~~~~~~~~~~~~~~~~~~

```

Changed severity_mid to uint8_t to match with nrf_log_internal.h like this for _std_0 - 6

```c
-void nrf_log_frontend_std_0(uint32_t            severity_mid,
+void nrf_log_frontend_std_0(uint8_t            severity_mid,
```

but no clue how to fix/change this

```c
targets/nrf5x/bluetooth.c:961:1: error: unknown type name 'nrf_log_module_dynamic_data_t'
 nrf_log_module_dynamic_data_t NRF_LOG_MODULE_DATA_DYNAMIC = {
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
targets/nrf5x/bluetooth.c:962:5: error: field name not in record or union initializer
     .module_id = 0
     ^
targets/nrf5x/bluetooth.c:962:5: note: (near initialization for 'NRF_LOG_MODULE_DATA_DYNAMIC')
```

https://github.com/espruino/Espruino/blob/c23289e6c18f9a88df812b855d2a54a000f54d61/targets/nrf5x/bluetooth.c#L961-L963

Any hints?

MaBecker, 2021-09-29:
Just tested with a SDK15 board and those need `severity_mid` as `uint32_t`. 

So at the moment there is no logging available for devices using SDK 12.

MaBecker, 2021-09-30:
After using a modified version of ble_uart it is now possible to run a test like `connect - send command - disconnect` for **_several hours without any error_**.  The ble_uart_Int is not using `NRF.requestDevice()`

This is the test snipped I uses:

```js
// device_a_b.js
/*

VERSION: "2v10.31",

A PuckJS f5b9 - "f0:d8:ef:29:f5:b9"  TX D6  
B PuckJS fed2 - "e2:9d:62:f9:fe:d2"  

1. MTU 23, stopped after 200 calls, 0 errors

2. MTU 53, stopped after 2921 calls, 0 errors

*/

Modules.addCached("ble_uart_Int", function() {
    exports.connect = function(id, ts) {
        var gatt, service, rx, tx, rxListener;
        var uart = {
            write: function(text) {
                return new Promise(function sender(resolve, reject) {
                    if (text.length) {
                        tx.writeValue(text.substr(0, 20)).then(function() {
                            sender(resolve, reject);
                        }).catch(reject);
                        text = text.substr(20);
                    } else resolve();
                });
            },
            disconnect: function() {
                return gatt.disconnect();
            }
        };

        return NRF.connect(id, { timeout: ts }).then(function(g) {
            gatt = g;
            // gatt.on('passkeyRequest', function() {gatt.sendPasskey(k);});

            return gatt.getPrimaryService("6e400001-b5a3-f393-e0a9-e50e24dcca9e");
        }).then(function(s) {
            service = s;
            return service.getCharacteristic("6e400002-b5a3-f393-e0a9-e50e24dcca9e");
        }).then(function(c) {
            tx = c;
            return service.getCharacteristic("6e400003-b5a3-f393-e0a9-e50e24dcca9e");
        }).then(function(c) {
            rx = c;
            rx.on('characteristicvaluechanged', function(event) {
                var d = E.toString(event.target.value.buffer);
                if (rxListener) rxListener(d);
                else uart.emit("data", d);
            });
            return rx.startNotifications();
        }).then(function() {
            return uart;
        });
    };
});

bleSendV2 = function() {

    require('ble_uart_Int').connect('e2:9d:62:f9:fe:d2 random', 3E3).then(function(uart) {
        uart.on('data', function(d) {});
        uart.write('digitalPulse(LED,1,200);\n').then(() => {
            c++;
            setTimeout(() => { uart.disconnect();
                print(c,e); }, 500);
        }).catch((e) => { print(e); e++;});
    }).catch((e) => { print(e); e++; });

};

bleAdvBat = function() {
    NRF.setAdvertising({}, {
        manufacturer: 0x0590,
        manufacturerData: JSON.stringify({ "bat": parseFloat(NRF.getBattery().toFixed(2)) })
    });
};

var c = 0, e = 0;

var BLE = LED3;

onInit = function() {

    // device A
    if (NRF.getAddress() == "f0:d8:ef:29:f5:b9") {
        print("device A");
        bleSendV2();
        setInterval(() => { bleSendV2(); }, 10E3);
    }

    // device B
    if (NRF.getAddress() == "e2:9d:62:f9:fe:d2") {
        print("device B");
        NRF.on('connect', function(addr) { BLE.toggle(); });
        NRF.on('disconnect', function(addr) { BLE.toggle(); });
        bleAdvBat();
        setInterval(() => { bleAdvBat(); }, 60E3);
    }
};

setTimeout(() => { save(); }, 2E3);

```

Will do one more mass test with nrf52840 devices, MTU 153 and larger data transfer, but I am pretty sure that this wir work too.

gfwilliams, 2021-09-30:
Would the change that is applied to SDK15 help here? `targetlibs/nrf5x_15/patches/0017-simple-log.patch`

Interesting - so you think it might be related to requestDevice? I guess there's a possibility that occasionally it might try to connect twice to the same device??

MaBecker, 2021-10-01:
The last test with  `NRF52840 with MTU 153 and larger data transfer` work without issues.

* `A`: using `ble_uart_Int`, with slice size 20 
* `B`: removed E.reboot()

> Would the change that is applied to SDK15 help here? targetlibs/nrf5x_15/patches/0017-simple-log.patch
> Interesting - so you think it might be related to requestDevice? I guess there's a possibility that occasionally it might try to connect twice to the same device??

Not sure, very busy with testing. Next test is to increase the slice size used for transfer  from 20 to a value matching the lager mtu size, starting with 75.

MaBecker, 2021-10-01:
I think it is definitive related to requestDevice!

> I guess there's a possibility that occasionally it might try to connect twice to the same device?

That's what I think too.

MaBecker, 2021-10-03:
**Testing now on NRF52840 mit MTU size 153**
 
Increasing `ble_uart_Int` slice eg to 140 byte  causes always `ERR 0xc (DATA_SIZE)` and no data is send, running with 120 byte cause no issue 

Looks like  there are some functions that can not handle larger data packages for now.

gfwilliams, 2023-05-17:
I think the original issue (advertising stopped) is fixed now - I fixed some issue with a race condition a few months ago.

Thoughts @MaBecker? can we close?

closed 2023-08-16 by gfwilliams