espruino/Espruino #2055: make SAVE_ON_FLASH more fine grained

fanoush, 2021-09-27:
Sometimes I need to cutdown normal nrf52 build by just few features currently removed now by `#ifndef SAVE_ON_FLASH`.

In other cases I want to start from SAVE_ON_FLASH configuration but want to add just few small extra features. I solved it by doing it like this:

rename current `#ifndef SAVE_ON_FLASH` code lines by more specific name e.g. `#ifndef SAVE_ON_FLASH_ARROWFUNC`

Then to cut down big build I define additional `-DSAVE_ON_FLASH_ARROWFUNC` in board file.
OTOH when done like that currently the feature gets included into SAVE_ON_FLASH build (for e.g. nrf51) which is also what I want.

to add this exclusion back to SAVE_ON_FLASH build it can be added to board file too, or there could be new section  in some common config header file with adding these automatically back to state as it was before like 
```
#ifdef SAVE_ON_FLASH
#define SAVE_ON_FLASH_ARROWFUNC
#define SAVE_ON_FLASH_NATIVE
#define SAVE_ON_FLASH_STORAGE
#define SAVE_ON_FLASH_FLASH
#define SAVE_ON_FLASH_SWSERIAL
#define SAVE_ON_FLASH_SWSPI
#define SAVE_ON_FLASH_SWI2C
#define SAVE_ON_FLASH_WATCHDOG
#define SAVE_ON_FLASH_DUMP
#define SAVE_ON_FLASH_WAVEFORM
#define SAVE_ON_FLASH_ERRORMSG
#define SAVE_ON_FLASH_RANDOM
#define SAVE_ON_FLASH_DATAVIEW
#endif
```
I don't have this header now.

It may not make sense to replace every SAVE_ON_FLASH ifdef, these above are just some that proved to be useful as a single feature to add e.g. to nrf51 build while still keeping size down or remove from nrf52 build to give more internal storage or cut it down (to e.g. fit dual banked bootloader).

It also helps as the name then explain what this feature is about so in future new code could be ifdef-ed like this instead of using generic   SAVE_ON_FLASH

Would you be interested in PR like this? I am currently keeping it for myself and have already features splitted  like list above.

If there would be such header that automatically puts stuff back then the (e.g. nrf51) board file would need to have some override like -DNO_SOF_NATIVE, -DNO_SOF_WATCHDOG to add it back to build. Or one could simply go and modify that system header and comment out some of the `#define` line mentioned above.

Also not sure how it affects autogenerated comments, some examples of this below
```
 /*JSON{
   "type" : "staticmethod",
-  "ifndef" : "SAVE_ON_FLASH",
+  "ifndef" : "SAVE_ON_FLASH_FFT",
   "class" : "E",
   "name" : "FFT",
   "generate" : "jswrap_espruino_FFT",
@@ -562,7 +562,7 @@ void jswrap_espruino_FFT(JsVar *arrReal, JsVar *arrImag, bool inverse) {

 /*JSON{
   "type" : "staticmethod",
-  "ifndef" : "SAVE_ON_FLASH",
+  "ifndef" : "SAVE_ON_FLASH_WATCHDOG",
   "class" : "E",
   "name" : "enableWatchdog",
   "generate" : "jswrap_espruino_enableWatchdog",
@@ -607,7 +607,7 @@ void jswrap_espruino_enableWatchdog(JsVarFloat time, JsVar *isAuto) {

 /*JSON{
   "type" : "staticmethod",
-  "ifndef" : "SAVE_ON_FLASH",
+  "ifndef" : "SAVE_ON_FLASH_WATCHDOG",
   "class" : "E",
   "name" : "kickWatchdog",
   "generate" : "jswrap_espruino_kickWatchdog"
@@ -1141,6 +1141,7 @@ void jswrap_espruino_dumpFreeList() {

 /*JSON{
   "type" : "staticmethod",
+  "ifndef" : "SAVE_ON_FLASH_DUMP",
   "class" : "E",
   "name" : "dumpFragmentation",
   "generate" : "jswrap_e_dumpFragmentation"
@@ -1178,7 +1179,7 @@ void jswrap_e_dumpFragmentation() {

```

fanoush, 2021-09-27:
I can probably put it to some branch so it can be seen how it looks.

gfwilliams, 2021-09-27:
Hi! This is something I have been meaning to do for a while, so a PR would be great.

In the `SAVE_ON_FLASH_ARROWFUNC` case, I've been trying to create new `#defines` prefixed with `ESPR_` to make it obvious they are Espruino-specific and have been trying (probably not well) to document them at https://github.com/espruino/Espruino/blob/master/README_BuildProcess.md#infomakefile-definitions

So maybe something like `ESPR_NO_ARROWFUNC` and then that `ifdef SAVE_ON_FLASH` maybe in `jsutils.h` for now? If we used `#if ESPR_NO_ARROWFUNC` then at least you might have the option of overriding one way or the other (enabling when disabled of disabling when enabled?).

For `SAVE_ON_FLASH_DUMP`/etc there's @jumjum123's Blacklist file which might be more flexible? https://github.com/espruino/Espruino/blob/master/Makefile#L47-L49

And code that handles it is at https://github.com/espruino/Espruino/blob/master/scripts/build_jswrapper.py#L216-L269

I think where it's just functions that are to be removed, it probably makes more sense to use that. Maybe the best thing is to add another line after:

```
if 'BLACKLIST' in os.environ:
	jsondatas = removeBlacklistForWrapper(os.environ['BLACKLIST'],jsondatas)
```

which is more like:

```
if 'BLACKLIST' in board:
	jsondatas = removeBlacklistForWrapper(board['BLACKLIST'],jsondatas)
```

And then you can specifically name functions that you don't want to appear in builds from the board file itself?

Maybe the thing to do is to just define a generic Blacklist file that we use instead of the `SAVE_ON_FLASH` defines as well? Not sure what you think there

fanoush, 2021-09-29:
Good tip about the black list, was wondering what it is - it is used in MICROBIT1 to remove OneWire. 
However if blacklist would replace most of SAVE_ON_FLASH definitions how would you  get the documentation properly generated for the reference site? Currently the `"ifndef" : "SAVE_ON_FLASH_WATCHDOG",` (or with new naming `"ifndef" : "ESPR_NO_WATCHDOG",`) inside json descriptions helps with that as you could include comment - this feature is not available when XX?

Also does it prevent C compilation of the unused code, probably not? It is only for linker LTO to optimize unreachable code out?

Anyway, the blacklist is only about removing specific jswrapper methods completely. It does not solve #ifdefs inside the code. I would prefer having one method for both for simplicity. But still the blacklist is probably neat for someone who does not want to touch the source code at all.

Naming ESPR_NO_XX instead SAVE_ON_FLASH_XX is a bit shorter, I'm for it :-)

gfwilliams, 2021-09-30:
Yes, documentation would be an issue I guess, but we could probably hack something up. Moving to ESPR_NO_FOOBAR the comments in the docs are basically going to end up being "Foobar: this isn't available on devices without foobars" which isn't that helpful anyway. I guess it'll end up being "This isn't available on some devices", which is better than nothing.

You're reliant on the linker to pull it out for sure, *but* the plus side of that is you're always sure the code can be compiled properly :)

But yeah, maybe stick with ESPR_NO_XX for now

gfwilliams, 2023-09-08:
Closing as complete - list of SAVE_ON_FLASH predefines is in https://github.com/espruino/Espruino/blob/master/src/jsutils.h#L41 and they get documented in https://github.com/espruino/Espruino/blob/master/README_BuildProcess.md#infomakefile-definitions

Although there will always be more cases where we could allow stuff to be compiled out :)