espruino/Espruino #2060: recompute compressed size when too big to save?

fanoush, 2021-09-29:
https://github.com/espruino/Espruino/blob/c23289e6c18f9a88df812b855d2a54a000f54d61/src/jsflash.c#L832

Shouldn't this code recompute compressed size after trying hard to free stuff few lines above? The variable still have old size so will probably fail second time again. or maybe I'm missing something?

BTW, was checking the code when trying to remove save() feature completely. It saves 2KB  :-) also code running directly from flash (.bootXX files) saves RAM. Loading .varimg fills the RAM - probably not great on nrf51

gfwilliams, 2021-09-29:
Wow, yes - thanks! It should run 'COMPRESS' again.

> was checking the code when trying to remove save() feature completely. It saves 2KB

That's a good point. It's definitely something that might be good to be able to remove.

On micro:bit I wonder if `save()` is a nice way to save the state of an existing app for kids, but then probably having more flash available for saving code is better :)

gfwilliams, 2021-12-15:
I think this is fixed now

closed 2021-12-15 by gfwilliams