espruino/Espruino #2061: idea - disable GPS uart between NMEA sequences

fanoush, 2021-09-30:
https://github.com/espruino/Espruino/blob/c23289e6c18f9a88df812b855d2a54a000f54d61/libs/banglejs/jswrap_bangle.c#L2126

just an idea, not sure how doable or practical - the nordic uart draws a lot (?)  would it make sense to disable it between NMEA sequences? once GPS device is powered on and initial messages are received it is pretty predictable when the next sequence comes based on refresh settings, would it save significant power to just turn uart off and wake it few (hundreds of) miliseconds before the message is expected and then after NMEA is received schedule next sleep precisely from that point again?

depends how much GPS draws in total, maybe it could save something significant during all day hiking trip?

gfwilliams, 2021-09-30:
Well, it's definitely possible. However I'm not really sure it'll be a noticeable help  - even in the lowest power modes it seems the GPS receiver draws something like 15mA constantly, so while we could shave 1mA off that, I doubt most people would notice :(

Having said that, having RTS/CTS handling internal to Espruino would be pretty cool and would provide an easy way to save power in a variety of different use cases (although I don't think RTS/CTS is connected in Bangle.js)

fanoush, 2021-09-30:
oh, that's too bad then, hoped it coul be like 1/3 or 1/4 but 1/15 is indeed not very significant, and when checking nordic devzone now without using dma it is even much less than 1mA https://devzone.nordicsemi.com/f/nordic-q-a/72535/52833-uart-power-consumption

closed 2021-09-30 by fanoush