espruino/Espruino #2090: Self-selecting menu items

stephenPspackman, 2021-11-05:
Bangle emulator 2v10.187

I've been having trouble using menus in the Bangle 2 emulator. I _think_ what is going on is that on the Bangle 2 the menu items are selected by touch-release events, but there is no logic to ensure that the press being released is ‘owned’ by the menu (and was not initiated before the menu was rendered). This means that if you use a touch event to open the menu, then whatever item is currently active is immediately selected. If the menu has been used previously, this may be the ‘menu close’ item, possibly giving the impression that the menu never triggered.

Note that (somewhat counterintuitively) this can bite even if the menu is triggered by touch-release, since the order of delivery of events is unspecified, so the same touch-release that opens the menu with one listener can close it with another.

The workaround seems to be to set the <menu>[''].selected = -1 in external code (not the predraw, which I'm not sure is even functional?) before opening the menu.

Reproduction: load this into the emulator, and click on the display surface. Note that an ‘!’ appears on the console instantly: item 0 has already been triggered on entry. Select item ‘<’. The menu collapses and an ‘x’ appears on the console, as expected. Now click the display surface again. The menu does not seem to appear, but another ‘x’ is printed: the menu was evidently activated and self-triggered its own close item!

```
const m = {
  '': {title: 'a'},
  '!': _ => print('!'),
  '<': _ => {print('x'); E.showMenu();}
};

Bangle.on('drag', e => !e.b && E.showMenu(m));
```

gfwilliams, 2021-11-10:
Thanks! I can reproduce this on real hardware too.

That definitely looks like what's happening. I'm surprised this hasn't been hit with existing apps, but maybe if you use `Bangle.setUI` for input it generally uses button up events (or 'tap' - which is probably generated on release too) then you don't have any problems

stephenPspackman, 2021-11-10:
Yes, I was attempting to put an additional menu of my own on
long-press-screen when I ran into this. The current workarounds are either
to gate my code with do-I-think-I-am-active flags, or (which is more fuss,
but probably more reliable) to deregister myself aggressively before
launching a menu.

Incidentally, a related thing that caused a lot of head scratching before I
figured it out is that the menu system does an LCD on, synchronously,
towards the end of rendering, so if you follow the advice of the tutorial
watch and put a watch face wakeup on LCD on, you get into a fight. I'm not
sure this is a bug, but it did confuse me for a few minutes and might be
worth mentioning in the tutorial (unless you did and I missed it!).

gfwilliams, 2022-06-08:
Closing this - I'm pretty sure it's fixed now by ensuring that setUI/showMenu all respond on the 'button up' event rather than down

closed 2022-06-08 by gfwilliams