espruino/Espruino #2092: How to compile the VM only?

paulrouget, 2021-11-08:
I'd like to embed espruino within my own project. But I don't need all the bluetooth or nfc features, or any non purely JS features. How to compile only the VM? And is there documentation of how to add it to my project?

Thank you :)

fanoush, 2021-11-09:
in that case you may check also other embeddable js engines, look at jerryscript (used by Kaluma project), you basically want to remove everything that makes Espruino unique, not worth it IMO

paulrouget, 2021-11-09:
Understood. Thank you.

closed 2021-11-09 by paulrouget

fanoush, 2021-11-09:
Really depends what you mean, just have a look at names of files here https://github.com/espruino/Espruino/tree/master/src
from what you said you want jslex,jsvar,jsparse maybe jsinteractive. Basically Espruino it is whole runtime including hardware support and set of API https://www.espruino.com/Reference , not just javascript interpreter. If you would want just remove NFC or Bluetooth but  still otherwise run Espruino runtime on nrf52,STM32, ESP32 (or Linux) then it may be worth it. If you really want just pure generic javascript interpreter which can be linked to your binary then it would be hard to separate. It may also depend on what "add it to my project" really means.