espruino/Espruino #2097: Bangle.js 2 ghosting

gfwilliams, 2021-11-17:
Right now I believe we flip the display refresh line on idle, but this means if the Bangle is busy the line doesn't get flipped and there can be ghosting (it seems especially if the display is cold).

We should flip the display line in the poll handler, being careful not to interfere with when the line also gets flipped when the display is updated

gfwilliams, 2021-11-17:
http://forum.espruino.com/conversations/369908/

dandelany, 2021-11-28:
I can confirm I see this issue sometimes too - nearly always when exiting an app & going back to the main screen. I can semi-reliably reproduce it by opening the app "Heart Rate Monitor", running it for ~30 seconds and then closing it. This app keeps the CPU pretty busy which seems to confirm your idle-related hypothesis.

Great job on Bangle 2, loving it so far :)

gfwilliams, 2021-11-29:
Just looking into this, and we do already update it in an IRQ, so it's not idle related!

Was the backlight on at the time? There are different update modes depending on if it's on - I'll need to look at it in more detail but it is possible that the 'backlight on' update mode stopped working when the Bangle reloaded a new app

dandelany, 2021-12-01:
Hmm... It seems that Heart Rate Monitor app keeps the backlight on, so it was on when I've noticed the "ghosting". But I still can't seem to reproduce it reliably. It's strangely inconsistent, both in occurrence and degree - sometimes (rarely) it's much worse than others, most of the time it's unnoticeable or barely noticeable. Will post again if i'm able to get it to happen more consistently.

gfwilliams, 2021-12-01:
Great - thanks! Yes, if you can find a way to reliably reproduce then it'll be easier to track down.

I did make some minor firmware changes to pulse width (with backlight off the pulse width was shorter than it was supposed to be) so it *may* help. We'll have to see.

halemmerich, 2021-12-09:
I can see the ghosting on both my bangles with firmware .219. Relatively consistent to reproduce with the QR code app and dark theme/watchface. Just display the QR code for about 30 seconds and switch to the watchface. I updated one of the bangles with .236 and so far it is looking good.

gfwilliams, 2021-12-10:
Ok, that's great news - so it's pretty easily reproducible with .219 but not with .236?

The tweaks I made *may* have fixed it, but it's also possible the burn-in is from the delay while loading a new app, and the new firmware is just substantially faster loading apps - so this is still worth keeping an eye on.

halemmerich, 2021-12-11:
I have done some more experiments:

- Both my bangles are now updated to .236, one still shows ghosting after showing a QR for about 30s and switching to clock and then the ghosting persists everywhere dark for some seconds. The other seems to be completely fine.
- I have done a little script to try and reproduce the behavior inside one app without loading something in between. Both bangles show no ghosting during this test.
- Behaviour is the same after some Minutes in the fridge ;)

The script: https://gist.github.com/halemmerich/e963d7cbbbd53a64f026d4f8ec9ea967

Is there something different during loading, that i could insert into my script to test it?

halemmerich, 2021-12-11:
In search for differences between the bangles, I have encountered this error on the black bangle with the ghosting.
```
Uncaught Error: Function not found!
 at line 13 col 667 in .boot0
...oad('welcome.app.js')})}})()(function(){var settings=require("Storage"...
                                        ^
>
```
After updating all apps on the black bangle to current state, it seems my previously reproducible steps for creating the ghosting no longer work.
- Restart with long press, hold until progressbar in boot screen is full
- Start script from gist with the web IDE
- Default settings produce ghosting
- BTN1 press turns LCD backlight off and on and seems to cure ghosting for the following cycles

halemmerich, 2021-12-11:
Flashing down to .222 restores ghosting. Reflashing to .236 still shows ghosting. Changing a setting which causes .boot0 to be rewritten prevents ghosting again on .236.

gfwilliams, 2021-12-13:
Thanks!

I think that `Function not found!` error was to do with the bootloader code on `.219` firmware. It would affect execution of subsequent stuff in `.boot0` but I don't think it was a big deal.

> it seems my previously reproducible steps for creating the ghosting no longer work.
> Flashing down to .222 restores ghosting. Reflashing to .236 still shows ghosting
> Changing a setting which causes .boot0 to be rewritten prevents ghosting again on .236.

So are you saying you think that maybe the JS that came with .222 caused a problem, and that when everything is updated to the latest (incl 236) it's all fixed? Or can you still reproduce it on 236?

gfwilliams, 2021-12-13:
Worth adding that when the backlight is on, the screen refreshes at about 120Hz which I guess clears ghosting pretty quick, so that's what you might see.

halemmerich, 2021-12-13:
Current state of bangles: Firmware 2v10.236
I used .222 because I couldn't find a .219 zip to restore my bangle to it's out of box state.

Preparation for similar starting point ghosting (probably not really necessary, it works with my daily driver blue bangle without resetting):
- Update settings app (to get factory reset)
- BTN1 long press for reload
- BTN1 short press for launcher
- Settings -> Utils -> Factory Reset -> BTN1 -> Yes

Reproduce ghosting:
- Power off (I use Bangle.off() in IDE terminal)
- Power on, upload gist via web IDE, **before** display turns off because of LCD timeout
- Let run one or two cycles of script, afterwards ghosting
- Touch the display, the backlight turns off for 200ms
- Residual ghosting for one cycle (maybe more depending on when display backlight was toggled and how many "ghosting" cycles ran before)
- At this stage I can't force new ghosting anywhere

Results:
- Ghosting happens on current firmware.
- Turning the backlight off fixes it until the bangle is rebooted (seemingly the cause for turning off the backlight is irrelevant, works manually or with configured timeout)
- Whatever the actual reason, it is not a real problem for usage anymore, because most bangles will turn their backlight off at least once relatively short after a reboot.

halemmerich, 2021-12-13:
> Flashing down to .222 restores ghosting. Reflashing to .236 still shows ghosting. Changing a setting which causes .boot0 to be rewritten prevents ghosting again on .236.

This can be explained by the needed reboot. Blue bangle wasn't reboot during testing. Has probably nothing to do with .boot0, just concidence.

halemmerich, 2021-12-13:
Flashed the black one back to .222, everything is really similar. The ghosting seems to be a little less pronounced on .236, but that could also be variances in my hardware, e.g. slightly different backlight brightness or color. The difference is small enough that I couldn't capture a good picture.

Izaic, 2021-12-16:
Getting ghosting on 2v11 as well

gfwilliams, 2022-02-01:
Please can someone have a check again? Last update was from a month ago and I haven't personally noticed issues on new firmwares

halemmerich, 2022-02-02:
I have seen it two times on 2v12, but i can not yet reliably reproduce.

jes, 2022-02-09:
I haven't got as far as a simple test case yet, but I can report that I'm seeing ghosting on my Bangle.js 2 while trying to update wpmoto for Bangle.js 2. I'm on 2v12.

elykittytee, 2022-02-19:
hello! I just received mine yesterday and noticed ghosting as I was playing around with it today. I am also on version 2v12.

gfwilliams, 2022-07-27:
No complaints in the last 6 months... Calling this closed

closed 2022-07-27 by gfwilliams