espruino/Espruino #2098: Phantom error in complex LHS expression in IDE.

stephenPspackman, 2021-11-19:
```
let a = {};
(a[1] || a[1] = {})[2] = 3;
```
The IDE reports “Bad assignment” on the second line, even though there's nothing wrong and the statement executes correctly:
```
>a
={
  1: { 2: 3 }
 }
```

stephenPspackman, 2021-11-19:
Another, simpler, and even more irritating example:
```
if (!i--) return;
```
Again, the IDE reports “Bad assignment.”

stephenPspackman, 2021-11-19:
Actually, I've made a mess—the two examples are not the same. The first example is correctly reported by the IDE and incorrectly accepted by the interpreter—I just noticed the missing parentheses (I obviously intended `(a[1] || (a[1] = {}))[2] = 3;)` and hallucinated that that was what I'd typed when the interpreter accepted it). The second example, however, is incorrectly reported by the IDE and correctly accepted by the interpreter.

Sorry, for that, I think I'm overtired.

gfwilliams, 2021-11-19:
It seems that the first example of `(a[1] || a[1] = {})[2] = 3;` does exactly the same thing in Espruino and in Node.js - the browser does report an error with it though (which seems odd to me as Node and the browser should both be using V8).

The IDE's linter is meant to report potential issues that may still be valid JS, so I wouldn't be too surprised, but `if (!i--)` is a bit frustrating. It's a totally off the shelf lint so I wonder why this hasn't come up before. I guess we could just disable the 'bad assignment' warning