espruino/Espruino #2102: Bangle.js 2 LCD backlight turns off after opening launcher

mabuchner, 2021-11-29:
Whenever I open the launcher (press button while on watch screen) on my Bangle.js 2, the LCD backlight turns off a split second after the launcher shows up.

I still can navigate in the menu, but the screen is really hard to see without the backlight at night.

I've already set the LCD Timeout setting to 60, but it doesn't seem to change anything.

I'm using firmware version 2v10.232.

mabuchner, 2021-11-29:
It might not be a problem with the launcher. It could just be a timing effect. The backlight seems to turn off after around 3 seconds and the screen gets locked after 5 seconds. The launcher seems to need around 4 seconds to start. Hence, I'm starting to interact right between backlight off and screen lock.

However, I would still assume, that the backlight turns on after I start to scroll up and down, but it doesn't.

mabuchner, 2021-11-29:
It looks like this was caused by the same issue as #2103 (a failed firmware update).

The "install default apps" in the web IDE solved the problem for me.

closed 2021-11-29 by mabuchner

gfwilliams, 2021-11-29:
Just a note that this could also be that you had a bunch of files on your Bangle which made it super slow (4s to load is a *lot*) and the backlight timed out while loading - so let me know if it happens again

gfwilliams, 2021-11-30:
Just to add:

> However, I would still assume, that the backlight turns on after I start to scroll up and down, but it doesn't.

I think that's a firmware issue that is solved in new firmwares

mabuchner, 2021-11-30:
> Just a note that this could also be that you had a bunch of files on your Bangle which made it super slow (4s to load is a lot) and the backlight timed out while loading - so let me know if it happens again

Yes, I have almost all Bangle.js 2 watch faces and applications installed 😄 

The launcher still loads approximately 4 seconds, but the backlight stays on.

mabuchner, 2021-11-30:
Yes, while I'm interacting with the screen, the backlight now stays on.