espruino/Espruino #2103: Bluetooth debug messages mess up clock screen

mabuchner, 2021-11-29:
Yesterday I've updated the applications and firmware (2v10.232) on my Bangle.js 2.

Since that time, I can see Bluetooth debug messages at the bottom of the screen, whenever Bangle.js connects or disconnect via Bluetooth.

[image]

As you can see, the debug messages also mess up the watch face.

I've double-checked and the Debug Info settings is set to "Hide".

gfwilliams, 2021-11-29:
Thanks - could you try just changing a setting in Settings, leaving the settings screen, then going back?

It might be that the '.boot0' file hasn't been updated properly.

mabuchner, 2021-11-29:
I changed the Beep setting. Went back to the clock screen. I then connected via Bluetooth and the debug message still showed up.

gfwilliams, 2021-11-29:
An 'install default apps' would fix it, but obviously that's not a great solution.

What happens if you connect with the web IDE and then long-press the button to reload the clock? Are any error messages shown?

mabuchner, 2021-11-29:
Unfortunately no error messages or warnings show up in the web IDE.

mabuchner, 2021-11-29:
I will try 'install default apps'.

mabuchner, 2021-11-29:
Yes, this actually fixed my problem.

Just FYI, I used the web IDE in Google Chrome on MacOS to perform the firmware update. The first time the update failed. Unfortunately, I did't note down any error messages. I just remember, that the loading bar in the browser didn't progress. The seconds time it appeared as if the update was performed successfully.

I will close this issue.

closed 2021-11-29 by mabuchner

gfwilliams, 2021-11-29:
Ok, thanks! Someone else reported something similar though too, so it is something I need to look at.

If you're able to figure out how to reproduce this please let me know! At some point I'll come up with a 'backup' tool that'll allow you to offload the contents of your Bangle in such a way that I could reproduce it here and debug it.

mabuchner, 2021-12-07:
It started to happen again. I updated my applications (including the boot loader) to the latest version.

After pressing the button to reload, all my settings were reset, I get the bluetooth debug messages and the backlight turns off really fast.

gfwilliams, 2021-12-07:
This sounds like this issue: https://github.com/espruino/Espruino/issues/2109

I'll try and get a fix in for it soon, but until then if you connect with the Web IDE and type `load("bootupdate.js")` in the left-hand pane (then long-press the button to reload the clock) I bet it'll be fixed?

mabuchner, 2021-12-07:
After running `load("bootupdate.js")` the watch got stuck at a "Reloading..." screen, but after a long-press of the button the problem went await again.

gfwilliams, 2021-12-07:
Yes, that's expected - as above you have to long-press to go back to clock. `bootupdate.js` isn't meant to be run on its own - but the command is handy to fix that issue for the mean time.

gfwilliams, 2021-12-07:
Ok, I've just hopefully fixed  #2109 so if you update to the latest Bangle.js firmware now, hopefully this will be fixed.