espruino/Espruino #2104: Scroll indicator in menu shows up inconsistently on Bangle.js 2

mabuchner, 2021-11-29:
I've noticed, that the scroll indicator at the bottom shows up inconsistently on my Bangle.js 2 (2v10.232).

For example, when opening the settings, then the bottom scroll indicator is initially visible

[image]

The scroll indicator stays visible until the "Theme" option

[image]

This kind of makes sense, since the list already arrived at the bottom.

However, when I scroll down to the next option, then the scroll indicator appears again.

[image]

When I scroll up from this position, then the scroll indicator appears also for the "Theme" setting

[image]

mabuchner, 2021-11-29:
(Random side note: The battery percentage in the battery widget increased although the watch isn't charging)

jes, 2022-01-27:
https://github.com/espruino/Espruino/blob/master/libs/js/banglejs/E_showMenu_Q3.js#L74

This is because it decides whether the arrow needs to be drawn based on whether or not the index 1 more than the number of the last drawn row is less than the number of menu items, but the `draw()` function takes arguments to specify the min and max row to update, to save it from having to redraw every row.

So when you only ask it to redraw 1 row, it thinks the last row that was drawn was not the last menu item, which means there must be more menu items, so it draws the arrow. I'll try and work on a sensible fix.

jes, 2022-01-27:
I've fixed it here: https://github.com/jes/Espruino/commit/5794dc3d270c5f0f49234b1484291076c72e338f

But it probably also wants to be fixed in the other 2 implementations of E.showMenu(), and I am struggling with working out how to minify it.

gfwilliams, 2022-01-28:
Great! There's an update_js.sh file in the parent directory there. If you have EspruinoDocs installed next to Espruino it should minify for you - but I'm very happy to do that. I'm just merging that change in now and will do changelog + minified JS

gfwilliams, 2022-01-28:
Thanks! Just added minified version with 1c2c11171b07bfa464f9d217b0413137c0eab370

A fix for the other implementations might be good *but* I'm not 100% sure if they have the problem - I'd have thought it'd have been mentioned by now if Bangle.js 1 had the issue.

jes, 2022-01-28:
Cool, thanks. FWIW the default Bangle.js 1 implementation does exhibit the same problem in the emulator. I guess most people just never noticed before!

jes, 2022-01-28:
For example, you can run this program in the emulator and press BTN1/BTN3 to scroll:

```E.showMenu({
  "": {"title": "Menu"},
  "one": function(){},
  "two": function(){},
  "three": function(){},
  "four": function(){},
  "five": function(){},
});```

gfwilliams, 2022-01-28:
Ok, great! If you could stick in a change for that too it that'd be really handy then! No worries about minification, it only takes me 30s

jes, 2022-01-28:
I've done that now for "F18" as well in https://github.com/jes/Espruino/commit/219d26fdb922352ea10e793f6b16437b69b80ed8. I also had a look in the "F5" version but it seems to draw the arrow unconditionally so I don't think any fix is needed there. I'm guessing F5 was some sort of development platform and isn't actually used in real watches?

commit 1c2c1117 (2022-01-28): Bangle.js2 : Fix inconsitent scroll indicator in menus (fix #2104)
commit a8c74cbe (2022-01-28): minify 2621fadb082f5fa7633744afa0ec47f995709cfb
Bangle.js1/2 : Fix inconsitent scroll indicator in menus (fix #2104)
closed 2022-01-28 by gfwilliams