espruino/Espruino #2108: Menu predraw does not seem to be called.

stephenPspackman, 2021-12-07:
My understanding from http://www.espruino.com/graphical_menu is that I should be able to

```E.showMenu({'': {predraw: g => g.setFontCustom(…)}, …}, …})```

and get a menu rendered in my own font (say, one with Latin-1 support) but this (or any other use of `predraw: g => …`) does not seem to work for me.

gfwilliams, 2021-12-07:
Thanks - yes, it looks like the `predraw` functionality was removed in the Bangle-specific version of the menus to make them update a bit faster. I guess it could be re-added.

However you mention Latin-1 - could it be you're seeing this problem? https://github.com/espruino/Espruino/issues/2105

It's on my list to fix sharpish, but honestly I'm spending over half the day every day just answering emails now :(

myxor, 2021-12-08:
I just found the same thing, wanted to go here to open an issue and saw this one.

Here is some example code which can be used to quickly reproduce this issue in the [WebIDE](https://www.espruino.com/ide/):

```
/*
 Example code for trying to set font of menu
*/

let fonts = g.getFonts();
var font = g.getFonts().includes("12x20") ? "12x20" : "6x8:2";

const mainmenu = {
    '': { 'title': 'Settings',  
          "predraw": function(gfx) { console.log("hello!"); gfx.setFont(font);} }, // not called
    '< Back': ()=>function() { console.log("back"); },
    'item1': ()=>function() { console.log("item1"); },
    'item2': ()=>function() { console.log("item2"); },
  };

E.showMenu(mainmenu);
```

closed 2021-12-08 by gfwilliams

stephenPspackman, 2021-12-14:
Should this be fixed in the emulator? Mine identifies as 2v10.187, and still does not seem to allow me to set a font.

gfwilliams, 2021-12-14:
Not yet - I haven't updated the emulator in some time

commit c1e2a603 (2021-12-08): Bangle.js: Fix E.showMenu clear if background color is set (https://github.com/espruino/BangleApps/issues/1024)
            Bangle.js: Re-add E.showMenu predraw callback (fix #2108)