espruino/Espruino #2109: Bangle.js 2: jshFlashWrite can be unreliable

gfwilliams, 2021-12-07:
See https://github.com/espruino/BangleApps/pull/986

Looks like sometimes a flash write on a 256 byte boundary can fail

closed 2021-12-07 by gfwilliams

Saduff, 2021-12-16:
Ran into this issue when updating apps today. Got a "jshFlashWrite failed" error during the upload of two apps and finally it got stuck on "uploading android" so I had to hold down the button to reload. I'm on 239 firmware where the patch should be included?

gfwilliams, 2021-12-16:
Well, there were a few commits around that time where there was actually a regression that made jshFlashWrite fail when it shouldn't (when writing <4 bytes) - so I imagine that's what you have. I bet updating to latest will fix it

commit fdd5b9bb (2021-12-07): Bangle.js: Add SPI flash read-after-write to verify write succeeded (fix #2109)