espruino/Espruino #2110: Time resets to time + timezone offset on app load

stephenPspackman, 2021-12-12:
Occasionally and intermittently, when loading a clock face on Bangle 2, I find that the time has been incremented by the time zone offset. I don't have a good characterisation of how or when this happens, but I'm reporting it in case others are seeing the same thing.

gfwilliams, 2021-12-13:
Ok, thanks - is it possible this corresponds with an upload via the Web IDE, or do you think it's purely when you upload an app via bangle.js.com/apps?

stephenPspackman, 2021-12-13:
I've certainly seen it from the web IDE, and I may even have seen it just
from a launch on the physical device, with no upload. I'll update next time
it happens to me.

gfwilliams, 2022-06-08:
Any news on this? Or shall I close the issue?

closed 2022-06-08 by gfwilliams