espruino/Espruino #2112: Data in memory is lost when bluetooth connection is interrupted

WizardStan, 2021-12-15:
### Summary
When the bluetooth connection is interrupted, step count and messages are lost.

### Environment
Bangle.JS 2; Up-to-date as of December 14th.
Android phone running Gadgetbridge.

### Steps to reproduce
1. Install Android app (v0.04), Messages app (v0.11), and Pedometer widget (v0.19).
2. Install Gadgetbridge on Android phone (v0.63.0).
3. Pair watch with phone through Gadgetbridge phone app.
4. Walk a few steps (10+ is sufficient) as noted by the Pedometer Widget.
5. Receive a notification (eg, send yourself an email) and verify it appears in the Messages app.
6. Turn off bluetooth on the phone; alternatively, set down phone and walk away, out of bluetooth range.
7. Take at least one additional step
8. Open the Messages app.

### Expected behaviour
Step counter continues to display the previous step count, plus 1 (or more, as more steps may have actually been taken).
Messages app displays previously received notifications.

### Actual behaviour
Step counter resets with next step, displays 1 (or so, depending on number of steps taken).
Messages app says "No Messages".

### Frequency
The notifications are lost the instant the bluetooth is disconnected 100% of the time.  The steps are less consistent, only sometimes (though frequently, about 3/4 of the time) lost when the bluetooth is disconnected.

### Additional notes
I suspect more data is lost than just the step count and notification list, but I don't have any other references to demonstrate the problem.
If the Messages app is open when bluetooth is disconnected, attempting to scroll the messages will produce graphical issues, and selecting a message will not do anything. "Back" still works to close the app.  Reopening the app presents the "No messages" message.

gfwilliams, 2021-12-15:
Which firmware version is on the Bangle? You say up to date - does that mean 2v10.248, or is it still the KickStarter firmware 2v10.219?

There was an issue where sometimes file writes would fail, and I think that is what you're hitting here. It's not related to the connection, it's just this issue that is already reported: https://github.com/espruino/BangleApps/issues/961

> Messages app displays previously received notifications.

This is expected. It always used to display previously received notifications and **then** someone filed a bug complaining that this was the case and saying that it should delete them - so now it does. Can't keep everyone happy :(

WizardStan, 2021-12-15:
> Which firmware version is on the Bangle?

2v10.248.  Not sure why I didn't just write that in the initial report.
If it's a write issue and unrelated to the connection then it's an unbelievable coincidence that I can reproduce it so often by turning off the bluetooth. Or at least I was able to yesterday, this morning I've gotten it to trigger once and haven't been able to since.  This was after having walked around for a bit, the daily reset had already happened, so I did the reproduction steps above: walked around a bit, turned off bluetooth, took a couple more steps, and noted it was back down to 3 from previous 140.
I'm also reading the source right now and the write happens during on('kill'), but this happens periodically throughout the day.  Unless I'm misunderstanding and on('kill') callbacks do get called (and the app gets unloaded from memory entirely) while the device is in use and not just as part of shutdown despite the description in the API.

> It always used to display previously received notifications and then someone filed a bug complaining that this was the case and saying that it should delete them

Whoever suggested that was absolutely wrong.  It should always be made as difficult as possible to delete data by accident, but if you walk just far enough away from your phone for the connection to drop and then come back it will be a surprise when the messages are gone.  Heck, my phone loses (and restores) the connection 4 or 5 times a day while it's in my pocket, hence my investigation and report, especially #2113 

> Can't keep everyone happy

Config options.
Now that I know this particular aspect was intentional and not a bug I can probably find time to create a pull request to add the option myself this weekend, if you want.

gfwilliams, 2021-12-15:
>  on('kill') callbacks do get called (and the app gets unloaded from memory entirely) while the device is in use

Yes, this happens every time you switch apps. So I guess after the step count resets it'd be good to see what's in the pedometer settings file. It might give a hint what's going on

> I can probably find time to create a pull request to add the option myself this weekend, if you want.

That'd be good, thanks! It's in `boot.js` in `android` - it'll be pretty easy to check the settings and see if you need to clear all messages or not.

WizardStan, 2021-12-15:
Since the step counter issue is actually unrelated to the bluetooth connection and already logged elsewhere, and the notification is by design, I'm going to close this.

closed 2021-12-15 by WizardStan