espruino/Espruino #2113: Notifications are not received if bluetooth connection is lost and then restored

WizardStan, 2021-12-15:
### Summary
When the bluetooth is unnaturally interrupted and then automatically restored, the watch stops receiving notifications.

### Environment
Bangle.JS 2; Up-to-date as of December 14th.
Android phone running Gadgetbridge.

### Steps to reproduce
1. Install Android app (v0.04) and Gadgetbridge Debug app (v0.01)
2. Install Gadgetbridge on Android phone (v0.63.0).
3. Pair watch with phone through Gadgetbridge phone app.
4. Open GB Debug on watch.
5. Receive a notification (eg, send yourself an email) and verify the expected JSON appears on the debug screen.
6. Walk away from the phone, out of bluetooth range, until the bluetooth is disconnected.  Verify connection has been lost in the Android app.
7. Walk back to the phone and wait for connection to be restored.  This may take a minute or two.
8. Open GB Debug on watch.
9. Receive a notification (eg, send yourself an email)

### Expected behaviour
New notification JSON appears on the debug screen.

### Actual behaviour
Debug screen continues to display "waiting for message".

### Frequency
100% reproducible.

### Additional notes
This has to be done when the bluetooth is unnaturally disconnected and automatically reconnected.  Manually turning the bluetooth off and on again does not produce this issue, and in fact doing so will serve as a workaround, restoring expected notification behaviour.
This may be related to #2112

johan-m-o, 2021-12-15:
Can confirm.

Bangle.js 2 2v10.247.
Gadgetbridge 0.63.0 on an Android 11 custom ROM (OnePlus 5T).
Android integration 0.05.
Messages 0.12.

It's not necessary to turn Bluetooth off and on to get it working again though. It's enough to disconnect and connect in the Gadgetbridge app.

gfwilliams, 2021-12-15:
More info on this forum post: http://forum.espruino.com/conversations/370937/

WizardStan, 2021-12-22:
I was just noticing that this hadn't happened to me in about a week now and thinking maybe the upgrade to gadgetbridge 0.63.1 had fixed it.  I have walked away from my phone on multiple occasions to have it disconnect and then returned to once again start receiving messages.  Even left it at home while I was out shopping yesterday and within a minute of returning home I got an email which displayed on my watch.  Seems I was no longer able to reproduce the problem at all.  So problem solved, I thought.

Today I upgraded to 2v11 (and upgraded all apps at the same time) and I'm back to being able to reproduce the problem 100%.  I have to turn bluetooth off and on again (or disconnect/reconnect through gadgetbridge) in order to get messages to start flowing again.  And now that steps are being properly tracked in both the watch widget and gadgetbridge phone app, I can confidently say that the communication failure is bidirectional: no steps are counted on the phone while in this connected-but-non-communicating state.

So this is to say, I upgraded to 2v11 and the problem persists, but it may not actually be as 100% reproducible as I'd originally thought.

gfwilliams, 2022-01-05:
Just a thought - but when you connected to Gadgetbridge did you choose to 'Pair' or 'Not Pair'. It's possible that the pairing info got lost upon update to 2v11 and that kicked off these issues...

WizardStan, 2022-01-05:
I do not remember, but it's back to working as expected and not able to reproduce the issue.

WizardStan, 2022-03-21:
I have not experienced this in months, at least not since upgrading to 2v12.  I'm going to close this under the assumption that something tangentially related was fixed that resulted in this getting fixed as well.

closed 2022-03-21 by WizardStan