espruino/Espruino #2116: new Date(str) incorrectly interprets ISO 8601 UTC timestamps

nebbishhacker, 2021-12-16:
`new Date(str)` incorrectly interprets ISO 8601 UTC timestamps as if they were in the local timezone.

On my Bangle.js, with timezone set to -5, this outputs `2000-01-01T05:00:00.000Z` when it should output `2000-01-01T00:00:00.000Z`:

```
console.log(new Date("2000-01-01T00:00:00.000Z").toISOString());
```

gfwilliams, 2021-12-16:
Interesting - thanks! So you're saying that `2000-01-01T05:00:00.000Z` is by definition in UTC time, so timezone shouldn't be added in that case?

rblakesley, 2021-12-16:
Yes, that's correct: https://en.wikipedia.org/wiki/ISO_8601#Coordinated_Universal_Time_(UTC)
I came across this a while back too, but then forgot about it...

nebbishhacker, 2021-12-16:
>  So you're saying that `2000-01-01T05:00:00.000Z` is by definition in UTC time

Yes - the `Z` indicates UTC time