espruino/Espruino #2119: Bangle.buzz may continue buzzing

gfwilliams, 2021-12-17:
I've had reports that Bangle.buzz can keep buzzing sometimes rather than stopping after a delay.

Any clues for how to reproduce this would be great.

gfwilliams, 2021-12-17:
Note: recent firmwares have now moved away from JS timers for handling the buzz (they happen in an IRQ) so what happens in JS *shouldn't* affect them

profriemke, 2021-12-20:
Maybe a way is described in the first post here: http://forum.espruino.com/conversations/371030/#comment16318046

profriemke, 2022-01-10:
After upgrade to espruino_2v11.9 I had to disable today all buzzes since they are now **_all_** not stopping. Even my previously  working notification integration now produces infinite buzzes.

gfwilliams, 2022-01-10:
Well that's odd. Interested to see what others find. For me it works fine - it's kind of hard to see how it could be failing in that way.

profriemke, 2022-01-18:
- solved