espruino/Espruino #2120: Bangle.js 2 Bangle.getPressure may not succeed on some devices

gfwilliams, 2021-12-17:
Reported at: https://github.com/espruino/BangleApps/pull/1092#issuecomment-996754700

It may work if the barometer has previously been powered on, but maybe it's related to the time allowed for the barometer to start up if it was previously off.

Doesn't happen on the first device I tried here so it may only be on some watches

elrod16, 2021-12-17:
You're on to something with the power up timing. It looks like SensiBLE works reliably because they explicitly power up the barometer and verify the power status before trying to fetch data from it. Or at least that's what I think it is doing.

elrod16, 2021-12-17:
Would it be practical to include some kind of non-busy waiting in the call? Like it sends the sensor power on, then idles until it gets returned a valid integer before returning a value to the calling app? I imagine though that there are plenty of ways that that could go wrong.

However, being it is a sensor that needs a brief "warm up", I imagine it would be practical for devs to expect a slight delay.

gfwilliams, 2022-01-04:
Thanks - `getPressure` returns a promise that resolves with the value rather than blocking, and yes, it does actually do a bit of a wait for power-up already (as far as I remember) so it *may* just be a matter of increasing that delay a little to account for some sensors.

I can do that and we can see if it fixes it for your Bangle however I'd like to try and find one here that it's an issue for, just in case the problem turns out to be something a bit tricker for fix :)

commit 5e2bf55d (2022-01-26): Bangle.js2: Increase wait after power-on for BMP280 to fix getPressure on some devices (fix #2120)
closed 2022-01-26 by gfwilliams