espruino/Espruino #2124: Bangle2 Serial1 handling for GPS AT6558 assumes UBLOX headers, needs to handle CASIC protocol

hughbarney, 2022-01-04:
Well bottomed out by @Fanoush and @Mark_M

https://github.com/espruino/Espruino/blo­b/master/libs/banglejs/jswrap_bangle.c
It checks for UBX at line 3690
The CASIC protocol headers bytes are 0xBA 0xCE.

Full thread at:
http://forum.espruino.com/conversations/371336/

There is a workround using Serial2 and poweron the GPS via D29.write(1);
@Mark_M has built a CASIC parser in the thread.

gfwilliams, 2022-01-04:
I've just moved this issue... But thanks! That's a pain.

So just to be clear -  I guess we have two options here?

* Disable the UBX handling for Bangle.js 2
* Replace the UBX handler with a CASIC handler for Bangle.js 2

I guess the second is preferable, we just have to convert the JS code in that thread to C

gfwilliams, 2022-01-11:
No reply so just doing the first as it's easier

closed 2022-01-11 by gfwilliams

fanoush, 2022-01-11:
not sure but does this change mean that CASIC data will be now available via GPS-raw event?
if not then reading Serial1 directly will still not work due to this line
https://github.com/espruino/Espruino/blob/bae246f6b53740a9f2ab2d11302fbb93d925ea98/libs/banglejs/jswrap_bangle.c#L3775 ?

gfwilliams, 2022-01-11:
Good point. You'll still get the GPS-raw but it'll be newline-terminated. It's probably not ideal but should be a definite improvement over what was happening before!