espruino/Espruino #2130: Generators

mikabytes, 2022-01-23:
Will [function*](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Statements/function*) come to Espruino? I checked the features list and it just had a dash `-`. Does that mean it's never going to happen, or just that it's not scheduled for work?

If not, I'm curious, is it a hardware restraint issue (makes binary too large or smth), or a matter of developer time/financing?

Generators can be very useful to save memory and clean up code. They're also necessary for some of the babel transpiling stuff, such as an `await` inside a loop. Of course, you can transpile without using generators, but my tests show the resulting code would be massive.

gfwilliams, 2022-01-24:
I wouldn't say never but it's not something on the todo list. There haven't been any real requests for it.

Looking at it, it seems to me that async/await (or the internals behind it) is required to make this work, since you've got to be able to stop execution of a function part-way through and then resume it later. And there's some discussion about that elsewhere but basically that is very hard to do with Espruino because of the way it interprets code...

Normally an interpreter would have compiled code down to either bytecode or native code which is kind of  'flat'. Once you've done that, stopping execution, storing the current position and execution context is relatively easy. Because Espruino is interpreting from source it's got to know not just where it was in the code, but that it was in a FOR loop, X way through iterating over some object.

So... I'd like to add async/await but it's a hard problem to solve, and I'm mainly lacking time for that. There are a lot more things that take priority for me (allowing Espruino to do things it couldn't before, rather than just making it slightly easier for devs to write code in a different way). Of course anyone else could contribute the functionality :)

Once async/await is done adding Generators on top should be reasonably trivial though.

mikabytes, 2022-01-25:
Cool, I think I understand the landscape a bit better now.

While we can definitely consider generators to be on my personal wish list, `async/await` would go even higher. 

We all lack time I suppose :) Would you say it's a very steep learning curve for contributing to Espruino? I might try taking a crack at it

gfwilliams, 2022-01-25:
> Would you say it's a very steep learning curve for contributing to Espruino?

I don't think so, no. You can just compile it to run on Linux, so especially if you're working on the language side of things it's pretty easy to iterate.

The interpreter is a bit different to most others - see http://www.espruino.com/Performance and http://www.espruino.com/Internals - so if you were used to working on other interpreters it could seem a bit confusing, but one big benefit is that you can just run it in a debugger and step through it (you're not having to work backwards though a YACC grammar or figure out what made a specific bit of bytecode)