espruino/Espruino #2137: `Bangle.getPressure()` fails sometimes with I2C Error on Bangle.js 2

malaire, 2022-01-30:
This code fails sometimes (I've seen failures after step 1, 19 and 20) with I2C error on Bangle.js 2. The "ERROR"-message from code is not output.

```
var counter = 0;

function foo() {
  Bangle.getPressure().then(d => {
    counter++;
    Terminal.println("" + counter + " " + d.temperature);
    setTimeout(function() { foo(); }, 10 * 1000);
  }).catch(error => {
    // This is never reached.
    Terminal.println("ERROR");
  });
}

foo();
```

Error is

```
Uncaught Error: I2C Error: Arbitration (start)
 at line 1 col 20
Bangle.getPressure().then(d=>{counter++;Terminal.println(""+co...
                     ^
in function "foo" called from line 1 col 5
foo();
    ^
in function called from system
```

gfwilliams, 2022-01-31:
What firmware version are you using? Just last week I made some changes to increase the time given for the sensor to power on, and it should fix issues like that

malaire, 2022-01-31:
I had 2v11. I've now updated firmware to latest available, "Firmware Update" -app says
> Your current firmware version is 2v11.67 and bootloader is 2v11.52

This seems much better, but I'm still getting that error rarely, so this has not been fully fixed.

malaire, 2022-01-31:
This failure is much more rare with 2v11.67 but I've already got it twice with 2v11.67 by re-sending the above code to Bangle.js 2 after it reaches about 20 steps.

gfwilliams, 2022-01-31:
Thanks - the problem is it's hardware-dependent, so on all the devices I have here it works fine.

Since the last PR made it better by increasing the delay, I guess I'll just increase it some more and hopefully that'll fix it.

gfwilliams, 2022-02-01:
I pushed an update yesterday which bumps the power up delay from 750ms to 1000ms - hopefully that'll fix it for you now?

malaire, 2022-02-01:
Still not fixed in 2v11.71 - I needed just 7 minutes of testing with the above code to get first error.

Also I don't understand how increasing power up delay could help here as I never get error on first `Bangle.getPressure()`. First call always succeeds and then later one of the subsequent calls fails.

gfwilliams, 2022-02-01:
> Also I don't understand how increasing power up delay could help here

Do you have other suggestions for how to fix this then?

It totally fixed the issue for another user with a similar problem, and the device I'm testing with here has been running for 24 hours without a problem.

malaire, 2022-02-01:
> Do you have other suggestions for how to fix this then?

No, I don't understand low-level details of Bangle.js 2 well enough to even understand why I2C error might happen.

The biggest problem here is that this error is **unrecoverable** - app may work well for an hour but once this error happens app stops receiving any temperature updates until it is manually restarted.

Would it be possible to make this error catchable or at least recoverable somehow?

malaire, 2022-02-01:
> The biggest problem here is that this error is **unrecoverable** - app may work well for an hour but once this error happens app stops receiving any temperature updates until it is manually restarted.

Looks like that problem has since been fixed. In 2v11 following code would stop after I2C error  but in 2v11.71 this continues on after I2C error.

```
var counter = 0;

function foo() {
  Bangle.getPressure().then(d => {
    counter++;
    Terminal.println("" + counter + " " + d.temperature);
  }).catch(error => {
    // This is never reached.
    Terminal.println("ERROR");
  });
}

setInterval(function() { foo(); }, 10 * 1000);
```

So while the actual problem hasn't been fixed, at least it doesn't stop temperature updates anymore.

gfwilliams, 2022-02-01:
Ahh - thanks. Sorry, I hadn't spotted that.

To make this keep working, all you actually need in your code is:

```
function foo() {
  Bangle.getPressure().then(d => {
    counter++;
    Terminal.println("" + counter + " " + d.temperature);
  }).catch(error => {
    // This is never reached.
    Terminal.println("ERROR");
  });
}

setInterval(foo, 10 * 1000);
```

So even if getPressure fails with an exception when it is called, execution continues.

However I'll push some changes in a second to make it reject the promise rather than firing the exception at the point getPressure is called. While it won't fix the I2C error it does make a lot more sense.

Also worth noting that if you really want to do this, you could just leave the Barometer on all the time with `setBarometerPower` and then respond to events from the barometer.

closed 2022-02-01 by gfwilliams

gfwilliams, 2022-02-01:
Yes, that's the one! Good point - I'll update the docs.

commit 79c2a6eb (2022-02-01): Bangle.js2: If errors do happen during Bangle.getPressure, reject the promise (fix #2137)