espruino/Espruino #2138: Problem with NRF.setScan() on ESP32

gurion2, 2022-01-31:
This simple function always halts with different reasons. Usually last received packet contains zero ID and impossible RSSI. Also filters (and another versions of SW) does not affect result - it always halts. 
```
NRF.setScan(function(d) {
    console.log(d); 
})
```

Some times I see **this** and function just stopts. 

```
BluetoothDevice: {
  "id": "00:00:00:00:00:00",
  "rssi": 394288,
  "data": new ArrayBuffer(0)
 }
```
or **this**
```
BluetoothDevice: {
  "id": "00:00:00:00:00:00",
  "rssi": 396336,
  "data": new Uint8Array([104, 64, 9, 128, 208, 99, 251, 63, 0, 0, 0, 0, 35, 12, 6, 0, 32, 12, 6, 0, 1, 0, 0, 0, 255, 255, 63, 179, 0, 32, 0, 0, 1, 21, 8, 128, 160, 99, 251, 63, 132, 64, 252, 63, 232, 101, 251, 63, 35, 12, 6, 0, 1, 0, 0, 0, 255, 255, 63, 179, 145, 253, 254, 63, 8, 0, 0, 0, 136, 0, 0, 0, 184, 55, 8, 64, 249, 255, 255, 63, 248, 255, 255, 63, 96, 0, 6, 0, 64, 0, 0, 0, 255, 255, 255, 255, 255, 255, 72, 157, 209, 179, 33, 22, 255, 255, 255, 255, 255, 255, 64, 43, 0, 0, 1, 8, 2, 4, 11, 12, 18, 22, 24, 36, 50, 4, 48, 72, 96, 108, 3, 1, 1, 45, 26, 111, 1, 23, 255, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 127, 8, 0, 0, 8, 0, 1, 0, 0, 64, 221, 7, 0, 80, 242, 8, 0, 17, 0, 19, 0, 254, 63]).buffer,
  "manufacturer": 65535,
  "manufacturerData": new Uint8Array([255, 64, 43, 0, 0, 1, 8, 2, 4, 11, 12, 18, 22, 24, 36, 50, 4, 48, 72, 96, 108, 3, 1, 1, 45, 26, 111, 1, 23, 255, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 127, 8, 0, 0, 8, 0, 1, 0, 0, 64, 221, 7, 0, 80, 242, 8, 0, 17, 0, 19, 0, 254, 63, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 252, 0, 0, 0, 0, 0, 0, 0, 16, 0, 0, 36, 0, 255, 64, 43, 0, 0, 1, 8, 2, 4, 11, 12, 18, 22, 24, 36, 50, 4, 48, 72, 96, 108, 3, 1, 1, 45, 26, 111, 1, 23, 255, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 127, 8, 0, 0, 8, 0, 1, 0, 0, 64, 221, 7, 0, 80, 242, 8, 0, 17, 0, 19, 0, 254, 63, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 252, 0, 0, 0, 0, 0, 0, 0, 16, 0, 0, 36, 0, 255, 64, 43, 0, 0, 1, 8, 2, 4, 11, 12, 18, 22, 24, 36, 50, 4, 48, 72, 96, 108, 3, 1, 1, 45, 26, 111, 1, 23, 255, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 127, 8]).buffer
 }
```

Sometimes ESP32 reboots by WDT
```
>ASSERT_PARAM(131072 0), in rwble.c at line 230
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x400867f6  PS      : 0x00060534  A0      : 0x8008a4f4  A1      : 0x3ffc05b0
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054
A6      : 0x3ffc1074  A7      : 0x60008050  A8      : 0x800867f1  A9      : 0x3ffc0590
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40086729  LEND    : 0x40086730  LCOUNT  : 0x00000000
Core 0 was running in ISR context:
EPC1    : 0x400892bd  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400867f6
Backtrace: 0x400867f6:0x3ffc05b0 0x4008a4f1:0x3ffc05d0 0x4008b3e3:0x3ffc05f0 0x4008381d:0x3ffc0610 0x4008b5c2:0x3ffde380 0x400d9b5d:0x3ffde3a0 0x400d9c4e:0x3ffde3d0 0x400d9a1d:0x3ffde3f0 0x400d9c4e:0x3ffde420 0x400d9a1d:0x3ffde440 0x400d9c4e:0x3ffde470 0x400d9bb1:0x3ffde490 0x400d9c4e:0x3ffde4c0 0x4010fdcb:0x3ffde4e0 0x400e1747:0x3ffde550 0x400e5365:0x3ffde5d0 0x400e5d54:0x3ffde6a0 0x400e7690:0x3ffde6e0 0x400e7774:0x3ffde710 0x400e77df:0x3ffde750 0x400e968d:0x3ffde770 0x400e9825:0x3ffde820 0x400f88fc:0x3ffde840
Core 1 register dump:
PC      : 0x401d8c6a  PS      : 0x00060f34  A0      : 0x8009450c  A1      : 0x3ffd6f40
A2      : 0x00000020  A3      : 0x00000020  A4      : 0x3ffc3d80  A5      : 0x3ffc1150
A6      : 0x3ffc4088  A7      : 0x3ffc3e2c  A8      : 0x3ffc3dbc  A9      : 0x3ffd6f30
A10     : 0x00000000  A11     : 0x00060823  A12     : 0x00060820  A13     : 0x3ffd8370
A14     : 0x3ffc4084  A15     : 0x3ffd8484  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Backtrace: 0x401d8c6a:0x3ffd6f40 0x40094509:0x3ffd6f60
Rebooting...
ets Jul 29 2019 12:21:46
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QOUT, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:2668
load:0x40078000,len:7304
load:0x40080000,len:5312
entry 0x40080274
WARNING: check error not implemented yet:0
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v11.61 (c) 2021 G.Williams
```

gfwilliams, 2022-01-31:
I'm afraid Bluetooth support on ESP32 is pretty flakey right now. PRs would be hugely appreciated for it :)

gurion2, 2022-02-01:
Small update. Sometimes setScan() thread stopts with MEMORY_BUSY error...

```
....
....
  "serviceData": {
    "f000": new Uint8Array([1, 0, 30, 3, 126, 67, 204, 76, 253]).buffer
   }
 }

BluetoothDevice: {
  "id": "f8:f2:8e:29:86:de",
  "rssi": -63,
  "data": new Uint8Array([2, 1, 5, 12, 22, 0, 240, 1, 0, 31, 3, 121, 55, 246, 60, 210]).buffer,
  "serviceData": {
    "f000": new Uint8Array([1, 0, 31, 3, 121, 55, 246, 60, 210]).buffer
   }
 }
New interpreter error: MEMORY_BUSY
> 
```

JThobbyist, 2022-11-01:
This is now fixed

gfwilliams, 2022-11-02:
Excellent - thanks!

closed 2022-11-02 by gfwilliams

MaBecker, 2022-11-21:
This will still cause unexpected errors  if the number of scans isn't limited!

Check this sample:   http://www.espruino.com/Reference#l_NRF_setScan

```js
// called in an environmet with lots of ble devices,  reboots after 210 outputs

NRF.setScan(function(d) {
    console.log(d); 
})

.......

BluetoothDevice: {
"id": "28:11:a5:a5:37:ae",
"rssi": -80,
"data": new Uint8Array([2, 1, 26, 3, 3, 190, 254, 7, 255, 16, 3, 64, 10, 1, 16]).buffer,
"manufacturer": 784,
"manufacturerData": new Uint8Array([]).buffer
}
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x401e95d7  PS      : 0x00060630  A0      : 0x8011a087  A1      : 0x3ffe9cd0
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x00000030  A7      : 0x00000030  A8      : 0x3ffccdc4  A9      : 0x00000001
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000007  LBEG    : 0x4009230c  LEND    : 0x40092328  LCOUNT  : 0xffffffff
ELF file SHA256: 0000000000000000
Backtrace: 0x401e95d4:0x3ffe9cd0 0x4011a084:0x3ffe9cf0 0x4013ae1a:0x3ffe9d10 0x4011c133:0x3ffe9d40 0x40120051:0x3ffe9dc0 0x4011d589:0x3ffe9ed0 0x4011d685:0x3ffe9f00 0x4011d734:0x3ffe9f20 0x4011db3b:0x3ffe9f40 0x4011db4f:0x3ffe9f60 0x4011db5f:0x3ffe9f80 0x4011e83c:0x3ffe9fa0 0x4011edf4:0x3ffe9fc0 0x4011efd0:0x3ffe9fe0 0x401204f8:0x3ffea000 0x40120b4c:0x3ffea0c0 0x4014d0eb:0x3ffea100 0x4011c133:0x3ffea180 0x40120051:0x3ffea200 0x40120b4c:0x3ffea2e0 0x401224a4:0x3ffea320 0x40122570:0x3ffea350 0x401225db:0x3ffea390 0x401244a1:0x3ffea3b0 0x4012463d:0x3ffea460 0x40133d50:0x3ffea480
Rebooting...

```

gfwilliams, 2022-11-23:
Interesting, thanks! I guess that's some kind of buffer overflow then? 

Does:

```
NRF.setScan(function(d) {})
```

Fail for you? eg if it can handle the devices fast enough, does it still break?