espruino/Espruino #2141: Bangle 2: Heart Rate Sensor stops working at some point

dashavoo, 2022-02-06:
**Device:** Bangle 2 (Non-kickstarter).
**Problem:** After a while, the heart rate sensor stops working. Espruino thinks it is turned on.
**Workaround:** Long hold the button to reboot the watch.
**Firmware:** 2v12

## Long description:
I have the heart rate sensor set to `always` in the health tracking options (Health Tracking app is at v0.11).
When I checked this morning to see what the readings were overnight, I noticed that the heart rate monitor no longer seemed to be active (no green LEDs, there had been no readings since late evening).

I rebooted the watch (long hold on button) and it started working again. I noticed because the `hrm` app was stuck on "Please Wait".

In the IDE, it seemed the Bangle thought the sensor was on: 
```
>Bangle.isHRMOn()
=true
```
but there was never an event (as expected, the lights were not on). `>Bangle.on('HRM', (hrm) => {console.log(hrm)});` never resulted in anything logged.

After rebooting the watch, everything was working as usual again.

Let me know if there is anything I can do to get more information out next time it happens.

gfwilliams, 2022-02-07:
Hi, thanks! did you got your Bangle from shop.espruino.com in the last week or so? If so I imagine it's one of the new batch with the updated VC31B sensor.

This issue could be related to that. Does:

```
Bangle.on('HRM-raw', (hrm) => {console.log(hrm)});
```

return anything? That's the raw data from the sensor - it's possible it thought it wasn't against your skin, but for some reason hasn't been able to turn itself back on

dashavoo, 2022-02-07:
Yes I did. I will try that the next time it happens and report back.

dashavoo, 2022-02-07:
OK, thanks. I was trying to capture as much heart rate data as I could, but I've probably got more than enough for the time being so will switch as suggested.

makecs50, 2022-04-13:
But what if i want to control lights being turned on after detaching watch from wrist? I really would like to turn off that skin detection feature, because it turns off sensor when i remove watch from wrist, but when i put it back on my wrist, sensor is not responding and lights are not flashing. And `Bangle.isHRMOn()` keeps returning `true`. Can i somehow get the state of lights, so based on that i could atleast invoke
```
Bangle.setHRMPower(0);
Bangle.setHRMPower(1);
```

Because of that skin detection feature i can't migrate my clients from bangleJS1 watch to bangleJS2.

gfwilliams, 2022-04-19:
I think you mentioned:

> Can i somehow get the state of lights, so based on that i could atleast invoke

But deleted it? If you look in the HRM-raw event there should be something like an `isWorn` field I guess you could check.

I plan to get the main issue fixed soon though - this doesn't affect the KickStarter bangle.js 2 though - just the new ones with VC31B

commit e3bd639c (2022-04-22): Bangle.js2: Fix wear detection on latest Bangle.js 2 (VC31B variant) (fix #2141)
closed 2022-04-22 by gfwilliams