espruino/Espruino #2145: Bangle.js colour palette bug

jes, 2022-02-09:
I have found that creating a string or object before allocating colour palettes can break the colour palettes.

I also observed some strange behaviour where changing the names of variables also changes the colours.

I can't quite figure out what's going on.

Here's a video example: https://www.youtube.com/watch?v=KYaI0fRVgEg

And here's some code you can copy and paste to test it out: https://gist.github.com/jes/9d5f7de8fea56c36d3181e515216e808

gfwilliams, 2022-02-10:
Hi - have you tried this on a real Bangle.js device?

I just tried here and I don't see the issue on a real device, so this may be an issue with the emulator - my guess is emscripten is unable to handle unaligned accesses, so when you add another variable things move around in memory and despite the address being correct the read of that address then fails.

jes, 2022-02-10:
D'oh! You're right. It only happens in the emulator. If I get a chance today I will try and dig into it and work out if I can fix it.

gfwilliams, 2022-02-10:
Thanks! I think it's probably this: https://github.com/espruino/Espruino/blob/master/libs/graphics/jswrap_graphics.c#L403

There's another instance of palettePtr being used too. I don't know if maybe we can hint to GCC that the pointer may not be byte-aligned? Either that or you'll have to #ifdef EMSCRIPTEN and the access each of the 2 bytes manually and combine them

Steven684, 2022-09-21:
Hi, Gordon @gfwilliams ,
I think I walk into a similar bug...
If I call the below function draw_string in a loop (with variating x,y,angle), the color keeps changing between the specified color and one other color... (magenta -> cyan, green -> magenta, red -> blue). Only black stays black, but that is 'transparent'...

```
function getImg(G, col) {
  return {
    width:G.getWidth(),
    height:G.getHeight(),
    bpp:1,transparent:0,
    buffer:G.buffer,
    palette:new Uint16Array([0xFFFF, col]),
  };
}

function draw_string(text, x,y, angle) {
  var oldCol = g.getColor();
  var txtSize = g.stringMetrics(text);
  var gtxt = Graphics.createArrayBuffer(txtSize.width,txtSize.height,1,{msb:true}); // w,h,bpp
  var gtxtimg = getImg(gtxt, oldCol);

  gtxt.setColor(oldCol);
  gtxt.setBgColor(0);
  gtxt.setFont(g.getFont());
  gtxt.setFontAlign(-1, -1);
  gtxt.drawString(text,0,0);

  g.setColor(oldCol);
  g.drawImages([ {image:gtxtimg, x:x,y:y, rotate: angle, center: true}, ]);
  g.setFontAlign(0, 0);
  g.drawString(text,x,y-20);
  g.setColor(oldCol);
}

```

gfwilliams, 2022-09-22:
How do you call it in a loop? setInterval?

because it seems you use `getColor` (so the current gfx color). Any widget/clock/etc that draws inbetween times could change what that color is, so maybe thgat's what you're seeing?

Probably if you prefix draw_string with `g.reset()` to reset colors it'll be ok?

Also, you use `gtxt.setColor(oldCol);` but `gtxt` is one bit. So only the bottom bit of `oldCol` is used (whereas from `g` it's 16 bit)

Steven684, 2022-09-22:
Yes, there is a setInterval, but within that, I call it in a for-loop. I am trying to print the numbers of an analog watch.
[image]
Since it's running in the emulator, I don't expect any other widgets/clocks to interfere.
And the g.drawString at the end of draw_string always shows in the right color, so getColor is always returning the right, expected color (didn't use it in this screenshot).

I've tried every combination of bpp's and colors, and although the setting might be wrong ('picking the bottom bit'), I'd expect it to be consistently wrong...
Right now, all numbers are randomly flipping between green and blue during the call to g.drawImages.

gfwilliams, 2022-09-23:
Ahh, ok - thanks. Sorry - yes, you're right then - this is the same issue with the emulator palette alignment issue.

As it happens, 1bpp images use the default foreground color so you don't actually need to do what you've done with the palette. The following code should work fine - even in the emulator:

```
function getImg(G) {
  return {
    width:G.getWidth(),
    height:G.getHeight(),
    bpp:1,transparent:0,
    buffer:G.buffer
  };
}

function draw_string(text, x,y, angle) {
  var txtSize = g.stringMetrics(text);
  var gtxt = Graphics.createArrayBuffer(txtSize.width,txtSize.height,1,{msb:true}); // w,h,bpp
  var gtxtimg = getImg(gtxt);

  gtxt.setBgColor(0);
  gtxt.setFont(g.getFont());
  gtxt.setFontAlign(-1, -1);
  gtxt.drawString(text,0,0);

  g.drawImages([ {image:gtxtimg, x:x,y:y, rotate: angle, center: true}, ]);
  g.setFontAlign(0, 0);
  g.drawString(text,x,y-20);
}
// test
g.setColor("#f00").setFont("6x8:2");
for (var i=0;i<10;i++)
  draw_string(i, 20+20*i,20, i*0.1);
```

Also, you may be able to get things going a little faster by pre-allocating `gtxt` to an image that's as big as you expect the biggest text to be (then just middle-align the text you draw) - since allocating graphics *can* take a while in some cases

Steven684, 2022-09-23:
Thanks! You're a great help! 😃 🥳 

Getting rid of the palette solves the problem, even when using background-color 'black' !
[image]

Also thanks for the pre-allocating tip 👍

gfwilliams, 2022-09-23:
Great! That clock face looks really nice too - and it's a great idea rendering the text to an image and rotating it :)

gfwilliams, 2026-09-03:
I'm pretty sure we fixed this now.

closed 2026-09-03 by gfwilliams