espruino/Espruino #2146: Backlight turning on with motion when touch is enabled

f8lxninja, 2022-02-10:
So for some reason the backlight will turn on when motion is detected, even though face up and twist are turned off. The only options I have it to turn on are BTN1 and touch. If I flick the watch repeatedly or move my arm in a consistent way the backlight turns on without even touching the screen or button.

If, however, you turn the wake on touch option off the issue goes away. Is this by design, or is it a limitation of how touch is perceived on the watch? I have a Kickstarter bangle 2 on 2v12.

gfwilliams, 2022-02-10:
Hi,

This is by design. Basically we can't keep the touchscreen on all the time on Bangle.js 2 because it would flatten the battery in a few days, so to detect 'wake on touch' we use the accelerometer.

So it's actually detecting a physical 'tap' on the device, which is why flicking it turns the bangle on. When you trigger it by moving your arm, is it when you do a sharp movement? There are some settings to 'tune' the touch sensitivity but it's hard to get something that works in all cases.

Perhaps there should be a setting for 'wake on double-tap' which would a lot harder to do accidentally.

f8lxninja, 2022-02-10:
Sometimes it is a sharp movement, other times it is not. One example is washing my hands. It is seemingly random when it decides that I had moved enough and turns the light on. Would it be possible to have an option to unlock the screen but do not wake the backlight if it detects movement? I have not found any settings to tune the touch sensitivity, only twist.

gfwilliams, 2022-02-10:
What do you think about the double-tap idea?

Yes, sorry - the actual accelerometer setup isn't exposed via menus right now, but it could be done using the `Bangle.accelWr` command. I imagine there will always be some cases where it goes off accidentally, or it'll miss taps which would also be annoying.

Right now there isn't individual tuning of when to turn backlight on or unlock - but that could be added at some point

f8lxninja, 2022-02-10:
Double tap could work, if you can configure the delay between taps. What I would like more is pressing the button wakes the touchscreen but only turns the light on when you start touching the display, and stays on for the delay that you set in the settings once you're done.

gfwilliams, 2022-02-11:
The double-tap is handled by the accelerometer chip, so the delay can be configured, but again it's not super-pleasant to do.

It might be that one solution is just to create an app that lets you fine-tune the accelerometer's settings.

So I guess (and this has been asked before) the main issue is splitting the LCD backlight and LCD lock options up so you can choose what happens when.

Maybe `wake on touch` should go back to being based on touchscreen, and we need `wake on tap` and `wake on double-tap`

f8lxninja, 2022-02-11:
I feel that's a good compromise. I am totally okay with it waking only on button press or some other means that is enabled in the settings.

dmitrizagidulin, 2024-01-18:
Running into this issue as well, and fortunately happened on this issue, from search. I don't suppose it's on the roadmap?

gfwilliams, 2024-01-19:
@dmitrizagidulin ok, so your issue was just the  screen wakeups when you weren't expecting it?

I think on recent firmwares the default is for wake on twist to be off as well (before it was on).

jonasborn, 2024-12-18:
I still think wake in touch could be a nice feature.
As for now, I can understand the power consumption issues, but isn't the TOUCH_PIN_IRQ already used, so having the CST816S in standby shouldn't take that much power? Might be totally wrong here, just started to explore the code base and the disassembly of the banglejs 2.

gfwilliams, 2024-12-18:
We already use the IRQ pin to respond to touches - the handler is at https://github.com/espruino/Espruino/blob/master/libs/banglejs/jswrap_bangle.c#L1975-L2001

The issue was that there wasn't a bunch of publically available info on the touch controller, and what most people seemed to be doing at the time (including the original SMA Q3 firmware) was this: https://github.com/espruino/Espruino/blob/master/libs/banglejs/jswrap_bangle.c#L2875-L2885

Basically shutting it down when not needed and waking it up with reset. Because when running it used ~1.6mA all the time.

Looks like there's a bit more info now - eg https://github.com/fbiego/CST816S - so maybe there is a sleep mode you can use

You can write direct to the touch controller from JS with `Bangle.touchWr` so you could experiment, and if you can get it working such that it'll actually enter a sleep mode where it doesn't draw a bunch of power then I'd love to pull that functionality in (you may even just be able to expose it as an app)

jonasborn, 2024-12-18:
Thanks, that's awesome to hear, 
I am going to have a look at it and will inform it was successful. Thanks for the quick response

gfwilliams, 2024-12-18:
Great! I should add that as far as I know the CST816S actually has its own firmware internally, so it's possible that the firmware installed on the ones in the Bangle doesn't support those extra features - but it's worth trying!

Actually measuring power usage can be a pain, but I have Bangles here where I've split them open to measure power usage - so if you get some commands that appear to work, let me know and I can tell you if they have any effect.

jonasborn, 2024-12-19:
Short update:

It seems that changing 0xFA to something away from test mode breaks the touch handling for unknown reason, it might be a required acknowledgement or sth. else. Setting it to 0x80, or 0xD0 or 0xA1 all works, as long enTest is enabled.
When enTest 0x01->0x06 are still changing when I touch the screen, so thats something good I think.

(Used https://www.waveshare.com/w/upload/c/c2/CST816S_register_declaration.pdf)

Probable touchHandlerInternal is filtering out the non-Test events. As I only can read the last event still stored in the register using JS, I can say why.

Using 
``` Javascript
Bangle.touchWr(0xFE, 0x00);
```
to enable auto sleep after 2s does not break anything but enTest, I do not know if it works at all.

Here is an overview of the cips register in default state (no touch applied):

``` Javascript
Bangle.setLocked(false); //Otherwise everything is 0x00

function readAllRegisters() {
  const registers = [
    0x01, 0x02, 0x03, 0x04, 0x05, 0x06, // Basic gesture and position registers
    0xA7, 0xA8, 0xA9,                   // Chip information registers
    0xEC, 0xED, 0xEE, 0xEF,             // Gesture and motion configuration
    0xF0, 0xF1, 0xF2, 0xF3,             // Low power scan data
    0xF4, 0xF5, 0xF6, 0xF7,             // Low power scan configuration
    0xF8, 0xF9, 0xFA, 0xFB, 0xFC, 0xFD, 0xFE // Sleep, IRQ, and other controls
  ];

  registers.forEach(addr => {
    const value = Bangle.touchRd(addr);
    console.log(`Register 0x${addr.toString(16).padStart(2, '0')}: 0x${value.toString(16).padStart(2, '0')}`);
  });
}

readAllRegisters();
```

| Register Address | Value  | Description                                  | Meaning                                      |
|------------------|--------|----------------------------------------------|----------------------------------------------|
| 0x01             | 0xFF   | GestureID                                   | Invalid or undefined gesture                 |
| 0x02             | 0xFF   | FingerNum                                   | Invalid or undefined value                   |
| 0x03             | 0x00   | XposH (High 4 bits of X coordinate)          | No valid touch position detected             |
| 0x04             | 0x00   | XposL (Low 8 bits of X coordinate)           | No valid touch position detected             |
| 0x05             | 0x00   | YposH (High 4 bits of Y coordinate)          | No valid touch position detected             |
| 0x06             | 0x00   | YposL (Low 8 bits of Y coordinate)           | No valid touch position detected             |
| 0xA7             | 0xB6   | ChipID                                      | Valid CST816S chip                           |
| 0xA8             | 0x00   | ProjID                                      | Likely default or unused                     |
| 0xA9             | 0x02   | FwVersion                                   | Firmware version 2                           |
| 0xEC             | 0x00   | MotionMask                                  | No gestures enabled                          |
| 0xED             | 0x01   | IrqPluseWidth (0.1ms units)                 | 0.1ms pulse width                            |
| 0xEE             | 0x01   | NorScanPer (Normal scan period in 10ms units)| 10ms scan period                             |
| 0xEF             | 0x06   | MotionSlAngle (Gesture sliding angle)       | Angle defined for sliding gestures           |
| 0xF0             | 0x00   | LpScanRaw1H                                 | No low-power scan data                       |
| 0xF1             | 0x00   | LpScanRaw1L                                 | No low-power scan data                       |
| 0xF2             | 0x00   | LpScanRaw2H                                 | No low-power scan data                       |
| 0xF3             | 0x00   | LpScanRaw2L                                 | No low-power scan data                       |
| 0xF4             | 0x04   | LpAutoWakeTime (Minutes)                    | 4-minute auto recalibration interval         |
| 0xF5             | 0x30   | LpScanTH (Wake threshold)                   | Threshold sensitivity value                  |
| 0xF6             | 0x03   | LpScanWin (Scan range)                      | Maximum range for scan, more sensitivity     |
| 0xF7             | 0x07   | LpScanFreq (Low-power scan frequency)       | Default frequency value                      |
| 0xF8             | 0x28   | LpScanIdac (Low-power scan current)         | Default IDAC value                           |
| 0xF9             | 0x02   | AutoSleepTime (Seconds)                     | 2 seconds of inactivity before sleep         |
| 0xFA             | 0x71   | IrqCtl                                      | IRQ control with gesture and touch enabled   |
| 0xFB             | 0x00   | AutoReset                                   | Auto-reset disabled                          |
| 0xFC             | 0x00   | LongPressTime                               | Long press reset disabled                    |
| 0xFD             | 0x00   | IOCtl                                       | Default IO settings                          |
| 0xFE             | 0x01   | DisAutoSleep                                | Auto sleep is disabled                       |

And just to be sure, as in https://github.com/espruino/Espruino/blob/master/boards/BANGLEJS2.py, the PIN for IRQ is D36?
If so, the state is always 1, probably as the touchHandler is used to read from the chip, setting D36 to low again long before any other watch is notified in JS? Not so sure how the watch system is build but I have to assume that the state sometimes must be low:
``` Javascript
if (state) return; // only interested in when low
```

For my next step it looks like I have to fire up the whole dev env here and get the values directly :)

gfwilliams, 2024-12-19:
Yes, I think you're right that Bangle.js is servicing the interrupt as soon as it asserts itself so you can't read the values.

Best bet if you want to do a custom build is to just follow the instructions at https://github.com/espruino/Espruino/blob/master/README_Building.md#super-easy-github-method

For the Bangle you literally just fork the GitHub repo, make a change and it'll auto-build it in a few minutes for you!

jonasborn, 2025-01-09:
Short update here:

I was able to compile by my self and get the IRQ pins value. I tried some funny values for scan frequency and standby mode but was not able to detect any changes. The chip drives me crazy and it feels like I am missing something.
Next step is to get something to detect frequency changes (as I am unsure if my cheap oscilloscope is detecting the touch or the bluetooth connection) or try to get some code running in flight mode to check again. For now have to prepare for exams, will come back to this.

enaon, 2025-01-14:
@jonasborn I really hope you can figure this out, auto standby mode on the touch screen is something I would really like too.

gfwilliams, 2025-01-14:
Just testing here with a Bangle that I've modified to get power usage from and:

```
NRF.setConnectionInterval(50); // lower BLE power usage
Bangle.setOptions({lockTimeout:0}); // keep unlocked forever
Bangle.setLocked(false); // power on
// 1.79mA used
Bangle.touchWr(0xFE, 0x00);
// exactly the same
```

I've poked around with a lot of registers and honestly I'd hoped that even changing the scan interval when not sleeping with `Bangle.touchWr(0xEE,30)` would make some difference, but I can't detect anything! 

The 1.79mA really does seem to match the 1.6mA the datasheet says for it being in its normal non-sleep mode (it goes down to about 0.15mA when touchscreen is off), but it does seem that regardless of what the datasheet says, we're just not managing to get it into sleep mode.

If it's not firmware maybe something about the installation (like something capacitive behind) it making it think there is something near and it can't get into sleep but it's hard to know.

If there are any commands you'd like me to try and run I can do that for you and see if we can drop the power usage...

enaon, 2025-01-14:
> If it's not firmware maybe something about the installation (like something capacitive behind) it making it think there is something near and it can't get into sleep but it's hard to know.

The thing is, there are two actionn the auto standby mode does. One is entering auto standby by it self when the time et at 0xF9 expires, and the second is that it auto wakes the 816, even if one manually enters sleep using 0xE5=3 (or 0xA5 =3 on some 816's)

so, even if something prevents it from entering standby mode, it should be able to exit if entered manually.

gfwilliams, 2025-01-14:
But doesn't `0xE5=3` enter Sleep mode, which is different from standby?

[image]

So you can only get out of sleep with a reset, but it's standby that's the one that should work without a touch?

Sadly `Bangle.touchWr(0xA5,3)` doesn't do anything. On the other 816s you're playing with, what does the firmware version register return?

enaon, 2025-01-14:
> But doesn't `0xE5=3` enter Sleep mode, which is different from standby?

yes, they are different modes, but I have seen it before, the p22 watch nd the p8 watch both comes with 816 versions, but one enters standby using the A5, and the other using the E5. Just curiosity. Also, there are versions that use 0xA5/0xE5=5 not 3. If you can test that too if you have some time and the ppk2 that can see small changes in power, it would be super.

gfwilliams, 2025-01-14:
This is interesting...

```
>Bangle.touchRd(0xa7).toString(16)
="b6"
// https://esphome.io/api/cst816__touchscreen_8h_source
 static const uint8_t CST816S_CHIP_ID = 0xB4;
 static const uint8_t CST816D_CHIP_ID = 0xB6;
```

so it's a **CST816D**! 

I'm struggling to find datasheets or anything and I can't seem to translate https://admin.osptek.com/uploads/CST_816_D_V1_0_2_1b06dfb078.pdf but it does show :

[image]

Which given the post above looks ominous

gfwilliams, 2025-01-14:
This might be some help - they read a few more registers (interestingly there may be a pressure one after x and y?)

https://github.com/TekuSP/Nanoframework-Drivers/blob/master/Drivers/CST816D/CST816D.cs#L65-L73

And they use 0xA5=3 for sleep - it's just a shame it doesn't seem to work here

enaon, 2025-01-14:
> This is interesting...
> 
> ```
> >Bangle.touchRd(0xa7).toString(16)
> ="b6"

I checked mine, bith the bangle.js2 and the P* report 180, b4, so they must have changed it in the newest bangle's, but maybe this is not it. 

>Bangle.touchRd(0xa7).toString(16)
>="b4"

gfwilliams, 2025-01-14:
> I checked mine, both the bangle.js2 and the P8 report 180, b4, 

huh - I just checked 3 bangles here and they report 0xB6 - but they are all a little newer. You have a KickStarter one?

enaon, 2025-01-14:
> tried that - sadly no luck

it feels like there are two possible cases. 

1. the autostandby is not enabled in the firmware
2. everything works, but something auto wakes the screen, like you said. 

if you have a ppk2,  maybe case 2 could be visible? I mean, if you change 0xFE to 1, and then 0, maybe you could see a down spike every 2 seconds?

enaon, 2025-01-14:
> so it's a **CST816D**!
>
> I'm struggling to find datasheets or anything and I can't seem to translate https://admin.osptek.com/uploads/CST_816_D_V1_0_2_1b06dfb078.pdf but it does show : [...]

I run this on google translate, it indeed says that only dynamic and sleep modes are supported on the 816D :(

So maybe they didn't implement the standby on the 816S, and then just used 816D?  In any case, if this is true and most bangle.js 2 have the 816D now, it seems like a dead end.