espruino/Espruino #2147: Bangle.js 2 very slow

Saduff, 2022-02-11:
It takes ~12.5 seconds to open the launcher. Even the "Loading..." popup appears ~3 seconds after button press. And it takes ~7.5 seconds to return to the clock after long pressing the button.

About my bangle:

* Firmware 2v12 and bootloader 2v11.52
* 38 apps installed
* Using the "Desktop Launcher" instead of the default, but switching to default made no difference

gfwilliams, 2022-02-11:
Please can you post these kinds of issues on the forum? This one was asked just last week I think.

Since you're now on firmware 2v12, just go to `Settings -> Utils -> Compact Storage`. It may take a minute or so to complete, but after that I bet it'll solve all your issues.

Potentially this is something I might arrange to happen automatically at midnight if the Bangle detects that Storage is getting fragmented.

Saduff, 2022-02-11:
> Please can you post these kinds of issues on the forum? This one was asked just last week I think.

Noted. Sorry about that, I did search the issues here, but not the forum. I'll make sure to check the forum next time.

Compacting storage indeed fixed it. The launcher now loads in ~2 seconds. Thanks!

closed 2022-02-11 by Saduff