espruino/Espruino #2149: [E.showMenu(...)] Entries displayed wrong on deletion/change from within the menu

storm64, 2022-02-14:
### Affected hardware version
Bangle, Bangle 2, reproducible on Web IDE Emulators

### Your firmware version
2v12

### The bug
I'm using a function to display a menu with changing entries depending on an outer variable.
When changing the variable and calling the function from the console, the menu is displayed as expected.
If the function is called from a onchange action inside the menu, it gets confusing. The menu is redrawing with the old entries, but at the moment a changed entry is selected it is displayed correct. If the number of entries has been reduced the last old entry will still be visible but is not selectable.

For reproduction run the following code:

```
var deleteA = false;
function testMenu() {
  var test = {
    "": {title: "Test"},
    "delete A": {
      value: deleteA,
      onchange: v => {
        deleteA = v;
        testMenu();
      }
    },
    "A": undefined,
    "B": undefined
  };
  if (deleteA) delete test.A;
  E.showMenu(test);
}
testMenu();
```

| changed first entry: | moved cursor down: | changed first entry again: | moved cursor down: | 
| --- | --- | --- | --- |
| [image] | [image] | [image] | [image] |

gfwilliams, 2022-02-14:
Ahh, thanks - so the issue is that upon returning from calling `onchange`, the original menu is redrawn.

If you delay the addition of the new menu with setTimeout then it'll work fine:

```
var deleteA = false;
function testMenu() {
  var test = {
    "": {title: "Test"},
    "delete A": {
      value: deleteA,
      onchange: v => {
        setTimeout(function() {
          deleteA = v;
          testMenu();
        },0);
      }
    },
    "A": undefined,
    "B": undefined
  };
  if (deleteA) delete test.A;
  E.showMenu(test);
}
testMenu();
```

And I think that's what we do in other apps that have this issue.

What do you think the correct behaviour should be? Because we could:

* Just call `onchange` *after* we redraw with the new value (but if there was some bounds/sanity checking to be done in `onchange`, it'd then be too late)
* If `onchange` returned true, don't then redraw
* Try and detect if `showMenu` had been called during `onchange` and avoid the redraw if so (but then what if something else has been done in onchange, like just drawing to the screen?)

I'm inclined to maybe just do the first option. It's easy and worst case you can just queue another redraw

storm64, 2022-02-14:
As I now understand the process better that leads to this "issue", I don't think it needs to be changed neccessarily.
In fact it is quiet obvious that succeeding to `onchange` a redraw is needed and therefore the initial menu object is used.

In my opinion there is another option by adding a seperate function like `oncommit` that is called after the redraw, like in your preferred first option.
Additionally this function could be called when a numerical value is selected. This would allow us to do complex calculation only after the user entered its choice.

closed 2022-03-10 by storm64