espruino/Espruino #2151: removeListener while executing events stops subsequent listeners from executing

gfwilliams, 2022-02-15:
Eg the code below:

```
function test() {
  print("A");
  E.removeListener("foo",test);
  print("B");
}
E.on('foo', test);
E.on('foo', () => print("C"));
E.emit("foo");
```

Should print `A B C` but it only prints `A B`.

gfwilliams, 2022-02-15:
Noticed in https://github.com/espruino/BangleApps/issues/1445

storm64, 2022-02-15:
While fiddling around with the code I experienced the following behavior:
```
function test() {
  print("A");
  E.removeListener("foo", test);
  print("B");
}
E.on('foo', test);
E.on('foo', () => print("C"));
E.emit("foo");
```
Returns: ` A ` and ` B `
Continue with:
```
E.emit("foo");
```
Returns: ` C `

The second emit seems to work as expected: The test function is removed and the reminding listener is called.

storm64, 2022-02-16:
There might be some use cases where it could come in handy to prevent the subsequent listeners to get called.

For example the lightswitch widget would be more usable on all apps (with enabled widgets) if touching the widget area could be masked out on the used app.

Example code that masks the left 50px:
```
// example touch listener
Bangle.on("touch", print);

// mask function
function touchMask(button, xy) {
  // check for the area to mask the touch event
  if (xy.x < 50) {

    print("code to be executed for the masked area");

    print("stop event handling");

    // removing and readding the listener to mess up executions
    Bangle.removeListener("touch", touchMask);
    Bangle["#ontouch"] = [touchMask].concat(Bangle["#ontouch"]);
  }
}

// make sure the mask function gets called first
Bangle["#ontouch"] = [touchMask].concat(Bangle["#ontouch"]);
```

Maybe its not a bug, its a feature. :sweat_smile:

I just build this into the lightswitch widget ([as linked above](https://github.com/storm64/BangleApps/commit/c2c812667f44578215b400dc924360786ecf198d)) for testing and it works quiet nice.

gfwilliams, 2024-10-10:
Fixing this might help with the problem: https://github.com/espruino/Espruino/issues/2559

gfwilliams, 2024-10-10:
Actually that doesn't help, but it's an easy fix.

`E.stopEventPropagation()` got added to stop events passing through so we don't need any hacks now